EmBitz/EBlink #3: Hotplug with Embitz 1.1 and EBlink

EmBitz, 2019-11-06:
To connect a running target without stopping it, Hotplug, in Embitz 1.1 you have to:
- Check the "Don't reset after connect" at the debugger target settings
- Disable flash loading in the debugger GDB server settings

In Embitz 2.0 there will be a Hotplug menu which will connect to a running target without the need to change settings.

T-Duke, 2019-11-07:
Hi!
Is it possible to use EBlink on Raspberry& I mean RPI4 on latest raspbian.
Thank you.

EmBitz, 2019-11-07:
I considered it a couple of days ago when I was porting it to linux_x86. I was not sure how useful it would be as a remote standalone GDB hub so I skipped the idea. I will see how easy this can be done.

EmBitz, 2019-11-16:
> Hi!
> Is it possible to use EBlink on Raspberry? I mean RPI4 on latest raspbian.
> Thank you.

Hi,

I have a raspberry pi build added to github. 
Have fun!

T-Duke, 2019-11-17:
Thank you! I'll try it on my RPI4

EmBitz, 2019-11-18:
> Thank you! I'll try it on my RPI4

Hi, if you are using V2 versions then you probably couldn't connect. I changed the interface init sequence but only tested it on V3 interfaces. The rc19.11-7 should have no problems and is tested with STmicro and Silabs and with V2 and V3 interfaces.

Cheers

closed 2019-12-02 by EmBitz

Miinik, 2019-12-26:
> In Embitz 2.0

Is there an approximate release date version 2.0?
Thanks.

AVI-crak, 2019-12-27:
There is another option - the publication of the source Embitz 1.1 as a starting version.

EmBitz, 2019-12-27:
@Miinik Just as you, I'm very busy right now with a IoT project for a customer. I'm using Embitz now for more than a year in debug build and fixed a lot of small issues during that year. As soon as I have some time I will build a 2.0 version (the build tree is now in the middle of a GCC tool migration)

@AVI-crak, sure those sources are becoming available again. I'm not sure if I use my own gitea or github. But, as in the past, don't expect people to actively join this project. I made a lot of things tweakable by using scripts where possible, but till now, only Miinik toke the time to add something.
There are too many free IDE's already in the market I think.

Miinik, 2020-02-03:
You are probably working on a very large project.
When in November last year I saw the announcement of version 2.0, I checked the site every day (sometimes several times a day). But...
Sorry.

EmBitz, 2020-02-03:
Yip, I'm busy with a wireless payment platform with LTE modems. Very big project.