EmBitz/EBlink #5: EBlink and QtCreator

seniorandre, 2019-12-01:
Hi Gerard,
I have written connector for EBlink to QtCreator and now test one.
As gdb debugger eblink works, but if I change code and try rewrite flash then I get a error of flash programming.
I include gdb logs for OpenOCD and EBlink for compare them. Output log from QtCreator console also... I do not understand where the cause of the error ...
Andrey.
[baremetal_ocd.log](https://github.com/EmBitz/EBlink/files/3907944/baremetal_ocd.log)
[qtcreator_output_console.log](https://github.com/EmBitz/EBlink/files/3907945/qtcreator_output_console.log)
[baremetal_eblink.log](https://github.com/EmBitz/EBlink/files/3907946/baremetal_eblink.log)

EmBitz, 2019-12-01:
Hi Andrey,

Sorry, that's why I still using the rc status.  I'm still busy with streamlining the code and sometimes these glitches occur.

Thanks for the report and I updated the repro.
Cheers Gerard

--- a/scripts/stmicro/flash/stm32HW_fl.script
+++ b/scripts/stmicro/flash/stm32HW_fl.script
@@ -121,7 +121,7 @@ function flash_write(sector, address, buffer)
     result = targetApi.writeReg("R2", buffer.byteLength);   // Halfwords count to be written
 
     // Run flash loader
-    result = targetApi.execute(0x20000000, 1); if(result<0) return result;
+    result = targetApi.execute(0x20000000, true); if(result<0) return result;
 
     // Disable the program flag  PG in flash_CR
     itrfApi.writeMem32(0x40022010, 0); if(result<0) return result;

EmBitz, 2019-12-01:
BTW
I hope to release a manual with all functions available in scripting as soon as doxygen is ready. By using doxygen I hope to tackle these problems by making the manual leading.

seniorandre, 2019-12-01:
Thanks....
It works :)

closed 2019-12-02 by EmBitz

EmBitz, 2020-04-16:
Hi, I was hitting your web site explaining how to use EBlink with QT and there is an example how to use the monitor reset. This is wrong, the "monitor reset halt" is not a valid command for bare metal.

the options are: monitor reset <0..3>

0 = SYSTEM_RESET
1 = CORE_RESET
2 = JTAG_RESET (pin)
3 = USER_RESET (by script)

Also EBlink has a modified CLI structure since  version 2 to make it even smoother to integrate with tools.