EmBitz/EBlink #6: EBlink and EmBitz 1.11

Tiilk, 2019-12-01:
Hello!
How to use EBlink in EmBitz 1.11?
I searched, but did not find how to do it.
Thanks.

EmBitz, 2019-12-02:
[image]  

[image]

closed 2019-12-02 by EmBitz

EmBitz, 2019-12-02:
And put in the "after connect":

`load`

in order to load (flash) the program.

Miinik, 2019-12-10:
I added EBlink to list of debugger interfaces for EmBitz 1.11.

[image]

To add, you need to extract this archive https://www.dropbox.com/s/4xzul9eagrpakju/EBlink_Interface_for_EmBitz_1_11.zip
to a folder 

> ${EMBITZ}\share\EmBitz\debuggers\Interfaces\  

EBlink must be installed in the folder

> ${EMBITZ}\share\EBlink

**${EMBITZ}** this is the folder in which EmBitz is installed. Usually this 

> C:\Program Files (x86)\EmBitz\1.11\

EmBitz, 2019-12-11:
Great, thanks!

Miinik, 2019-12-18:
Unfortunately I do not have much free time...

EmBitz, 2019-12-18:
Thanks, pity.

Miinik, 2020-01-21:
Updated debug interface for [EBlink 2.0](https://github.com/EmBitz/EBlink/releases/tag/2.0)
https://www.dropbox.com/s/ylyhi1b826b5fq7/EBlink_2.0_Interface_for_EmBitz_1_11.zip