EmBitz/EBlink #7: Debug programs in memory

AVI-crak, 2019-12-02:
Please add a program debugging script in memory. This is the case when the download address is different from 0x80000000, and is 0x20000000. In this case, loading into RAM memory is required, without modifying FLASH.

EmBitz, 2019-12-02:
Do we need that? Is GDB not recognizing that  the text section is written in volatile RAM?

I think the only thing that should be changed is the reset strategy, load the PC and SP with the right values after the load. Look at the stm32-auto.script

`function  reset_post(resetType)
{
    local targetApi = :: TargetAPI(); // Our interface to the target class

    // We don't use this if the reset is a user_script type. In that
    // case, the user is responsible for the initialization.
	//
    if( resetType < USER_RESET)
    {
        // Set the Stack pointer according the Vector table entry
        itrfApi.readMem32(0x20000000);
        targetApi.writeReg("SP", itrfApi.value32);

        // Set the Program pointer according the Vector table entry
        itrfApi.readMem32(0x20000004);
        targetApi.writeReg("PC", itrfApi.value32);
    }

    return 0;
}`

AVI-crak, 2019-12-02:
No contact...
Details: Version ST-Link V2 - V2.J27.S6, 
at startup, the debugger window draws this:
//
EBlink version rcl9.11-7 by Gerard Zagema

Interface at 3:17 - <BUS>:<ADDR>
Interface type : STlink/U2 
STlink connect : Under reset 
Target voltage : 3.25U 
Interface speed: 4000KHz
Target detected: Cortex-M7 <r0pl> with FPv5_SP
HU breakpoints : 8
HU uatcbpoints : 4
Fault unwind : Enabled
STmicro device : 0x449
Detected FLASH size: 0x100000
Configured RAM size: 0x50000
GDB listening <0.0.0.0 @ 2331>
GDB connected
//

There is no access to registers, no access to memory, the window for reading registers looks cropped - without R0-15 and S0-31.

EmBitz, 2019-12-03:
I have the strong feeling that this is not EBlink related. With the old Embitz STlinkGDB only changing the PC and SP initialization was enough.  You could take a look at the interface.script at the stlink debuggers folder of Embitz.

The GDB server is just a bridge between CMSIS-DAP and GDB.

But as soon as I have some time I will check it myself, to be continued....

Miinik, 2019-12-03:
@AVI-crak It works for me debugging in memory.

> [image]

In file **stm32-auto.script** I added a function 

> function reset_user()
> {
>    local targetApi = :: TargetAPI(); // Our interface to the target class
>    
>    // Set the Stack pointer according the Vector table entry
>    itrfApi.readMem32(0x20000000);
>    targetApi.writeReg("SP", itrfApi.value32);
> 
>    // Set the Program pointer according the Vector table entry
>    itrfApi.readMem32(0x20000004);
>    targetApi.writeReg("PC", itrfApi.value32);
>    
>    return 0;
> }

and set reset type 3.

[image]

EmBitz, 2019-12-03:
@Miinik  chapeau!
Me myself hadn't thought of an user defined reset scenario, really good. 
Nice to see that someone gets the idea and the strength of squirrel scripting, even without a manual with the available functions in scripting.

closed 2019-12-03 by EmBitz

EmBitz, 2019-12-03:
Tip: 
Place in your hardfault handlers _brkpt instead of endless loop.
If an exception occurs, EBlink unwind will bring you (most likely) to the cause of the problem.

Miinik, 2019-12-06:
I found a problem while debugging in memory. 
See the picture.

[image]

Calling the function **SetSysClock** located at address **0x2000071c**, but there is no data in RAM. I looked at the MAP file. The function is really located at this address.
Using STLinkGDB.exe does not have this problem. I assume that EBlink does not load all the data into memory.

EmBitz, 2019-12-07:
Strange, to be honest, it is not eblink who is loading the RAM but it is GDB. EBlink is (should be) transparent  to the RAM. Which device is it and could you make a small project with this problem?

EmBitz, 2019-12-07:
Also, with -v7 you turn on full debug and tracing. If you are familiar with GDB server protocol then you can check if GDB is writing that region.

Miinik, 2019-12-07:
Tested on STM32F411CEU6. Project https://www.dropbox.com/s/064xufotj1trpnw/Test_F411.zip?dl=0

In the script stm32-auto.script should be a function reset_user() https://github.com/EmBitz/EBlink/issues/7#issuecomment-561169518

EmBitz, 2019-12-07:
I don't have the F411 so I used a F446 and I can't reproduce the fault here. Can you modify ram inside that region?

[image]

EmBitz, 2019-12-07:
Update: On a F4906VG, no problem.

Miinik, 2019-12-07:
On F407VG has a similar problem.

As it turned out, not all data is also written to flash memory.
I added the tool to EmBitz.

[image]

Command line. 

> -I stlink,swd -D stm32-auto -F verify,run,file="${PROJECT_DIR}${TARGET_OUTPUT_DIR}${TARGET_OUTPUT_BASENAME}.hex"

I erased the chip in the program STM32 ST-LINK Utility.
Launched the tool in EmBitz.
EBlink work log.

> Launching tool 'eblink': C:\Program Files (x86)\EmBitz\1.11\share\eblink\eblink.exe -I stlink,swd -D stm32-auto -F verify,run,file="D:\STM32\STM32F407\Test\bin\Release\Test.hex" (in C:\Program Files (x86)\EmBitz\1.11\share\eblink)
> stdout> 
> stdout> EBlink version 1.0-[4] by Gerard Zagema
> stdout> 
> stdout> Interface at  1:1 - <BUS>:<ADDR>
> stdout> Interface type : STlink/V2
> stdout> STlink connect : Under reset
> stdout> Target voltage : 3.28V
> stdout> Interface speed: 4000KHz
> stdout> Target detected: Cortex-M4 (r0p1) with FPv4_SP
> stdout> HW breakpoints : 6
> stdout> HW watchpoints : 4
> stdout> Fault unwind   : Enabled
> stdout> STmicro device : 0x413
> stdout> Detected FLASH : 0x100000
> stdout> Configured RAM : 0x30000
> stdout> Reset: system
> stdout> 
> stdout> HEX file: D:\STM32\STM32F407\Test\bin\Release\Test.hex
> stdout> 
> stderr> Target Exception!
> stderr> 
> stderr> 	[LR]:	0x200004E1
> stderr> 	[PC]:	0x20000000
> stderr> Flash verify failure!Flash verify failed!!!
> stdout> Flashing sector  0, address: 0x08000000, length: 0x077C
> stdout> Flash verify .. failed!
> Tool execution terminated with status -1

Comparison of the code in the chip and the hex file.

[image]

Writing to flash memory using the program STM32 ST-LINK Utility performed without errors.

EmBitz, 2019-12-07:
Yip, that was something different.
Please checkout the new stm32f7f4_fl.script.

Thanks.

P.s. still doesn't explain why you don't have that setSysClock in ram. That has nothing to do with flashing.

EmBitz, 2019-12-07:
P.s. You project is using debug as directory for obj and output for both debug_RAM and debug. 
It's better to put every target in his own object and output space to prevent mix up. I had problems during this project test. I put debug_RAM in side DebugRam folders.

EmBitz, 2019-12-07:
P.s. (2)

If you run from RAM you also have to relocate the vector table at startup on system init.

Miinik, 2019-12-07:
Debugging in flash memory is also with such a problem.

[image]

Firmware size 0x0998. Written to flash memory 0x0708.

[image]

I used different ST-Link and chips F407VG and F407ZE. The same problem.

EmBitz, 2019-12-07:
On a 407VG it runs here fine, no problems at all. 
P.s. I just uploaded a voltage aware flash script.

EmBitz, 2019-12-07:
First screenshot is from an empty device, second screen shot is new startup. EBlink doesn't find any differences in flash and GDB code.
[image]
[image]

EmBitz, 2019-12-07:
On a F429 also fine (id 0x419), I don't know what the difference is between our two setups.
[image]

EmBitz, 2019-12-07:
Did you get the latest script folder from github?

Miinik, 2019-12-07:
Thanks. Flash write issue resolved. 
But only when there is no **erase** parameter. When there is this option, an error appears.
> stderr> Target Exception!
> stderr> 
> stderr> 	[LR]:	0x200004E1
> stderr> 	[PC]:	0x20000000
> stderr> Error: Sector erase timeout

But there is still a problem with debugging in memory. 
Now I see all the data in chip memory, but very often windows with an error message appear and a transition to an address outside the firmware.
An animated picture shows this.

[image]

EmBitz, 2019-12-07:
Ok, new flash script in github with erase timeout solved. According spec this can take even up to 32 seconds for low voltage 8bit.
I will see if I can find what is going on with that code in RAM. My feeling is that it is not EBlink related because all that RAM stuff is transparent but you never know.

Miinik, 2019-12-07:
OK.
But the strange thing is that when using STLinkGDB, there is none of these problems.

EmBitz, 2019-12-07:
Yip that's strange. I agree.

EmBitz, 2019-12-07:
Ok, i found it. Will try to come tomorrow with an update.

EmBitz, 2019-12-07:
[eblink.zip](https://github.com/EmBitz/EBlink/files/3935423/eblink.zip)

EmBitz, 2019-12-07:
This is my reset sequence.

[image]

Miinik, 2019-12-07:
Now debugging in RAM works great.
Thanks!

EmBitz, 2021-06-03:
REMARK:
This is a old thread, the CLI parameters of EBlink have been changed!!!!!!!!!

E.g. -D is now -S and -D is for defining script symbols etc etc