EmBitz/EBlink #8: -exec-interrupt -> SIGTRAP

seniorandre, 2019-12-02:
Hi Gerard,

And me again... with QtCreator IDE
When I send to EBlink signal -exec-interrupt, then is returned a response SIGTRAP.
and Creator show dialog message SIGTRAP.
Another debaggers return SIGINT response.
Can you change it?

Andrey.

[image]
[image]

EmBitz, 2019-12-02:
Are you sure? It is now sending a "S05" on halt but I can't find any spec which stipulates that it should be different. What should it be according you?

EmBitz, 2019-12-02:
Why is QT Creator  showing a message box when it hits a breakpoint, is that normal behaviour?  I would expect that it just update the UI of the IDE with the new status. At least that's how I did it when I build Embitz.

seniorandre, 2019-12-02:
-exect-interrupt is sent when you press interrupt for exit from infinit while(1) for example, without breakpoint.
Open OCD returns on that situation SIGINT and Creator is stoped on any line into while without the dialog.

https://en.wikipedia.org/wiki/Signal_(IPC)#SIGINT

[image]

EmBitz, 2019-12-02:
Yes, I just see that OpenOCD returns a S02 but stlink is just returning S05.  If I recall correctly, J-link server is also returning S05. From an IDE perspective, it is not important if it's S05 or S02 because the reason is in the stop frame.  Why should an IDE do differently between a stop with S02 or S05?

I will look into this, if I don't need to change this then I prefer to keep it this way. Using S05 is just straight forward, on status change to halt->throw S05. Otehrwise I have to keep track what the reason is, and what if we have a race condition ( -exec-interrupt on breakpoint trap) ?

But if this should be S02 to make it more compatible then I will do it for sure.

EmBitz, 2019-12-02:
What I could do is using the NVIC  (dfsr) debug fault status register and if the HALTED flag is the only one set, return S02. I think that that will do the trick. If any of the other flags are also set, I will return S05. 

http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0337e/Cihcbadd.html

seniorandre, 2019-12-02:
Into code I have found setting. :)

[image]
[image]

seniorandre, 2019-12-02:
> . If any of the other flags are also set, I will return S05.

It will be right .

EmBitz, 2019-12-02:
Yip, that's make sense. That stop code shouldn't be that important for the IDE, the stop frame should be parsed anyway.

I keep this in mind and if others are complaining I will change this. I just did some test how that DFSR register is handling the stop reasons. I think I could do something meaningful with it if necessary.

EmBitz, 2019-12-02:
Hi,

Could you try this version?  I'm parsing the stop reason from the core. 

[eblink.zip](https://github.com/EmBitz/EBlink/files/3913087/eblink.zip)

seniorandre, 2019-12-02:
That version do not works right. I have spontaneous interrupt on start.
[image]

closed 2019-12-02 by EmBitz