EmBitz/EBlink #9: Interrupt call failed: svc 0x0

AVI-crak, 2019-12-03:
Hello again.
My code stops at the command "svc 0x0", the LR register shows the transition to the interrupt mode 0xFFFFFFED, while the PC register is outside the range 0xFFFFFFFE. Interrupts are identified and recorded. For them there is program code with real addresses from the interrupt table. If you manually change the PC to the interrupt address, then the code is executed and correctly exits to the main program.

Extras: very long loading of a data image, tens of minutes. The server is trying to allocate memory for the external memory section, it is large, but it is marked for reading.
[image]

EmBitz, 2019-12-03:
And why should this be a eblink problem? 
I did a OS project with task switching without  problems. Don't think this is eblink related.

closed 2019-12-03 by EmBitz

EmBitz, 2019-12-07:
AVI, there was a fault in the EBlink which could explain your behaviour. 
I will see if I can update it tomorrow, Sorry!