EmBitz/EBlink #10: scriptname and hostname at start

seniorandre, 2019-12-07:
Hello, Gerard,

First:
I continue to correct code for qt-creator .  Very not convenient that scriptname works only without extention.
- scriptname is accepted  as simple scriptname into script directory - ok
- scriptname is accepted  as absolute path + scriptname - ok
- scriptname is accepted  as relative path + scriptname -ok
but is not accepted with extention. Very difficult to do parse that with standart functions on crossplatform systems (convenient choose file and save it with path or w/o path).

Second:
Why is default host adress - 0.0.0.0, that address is not valid. But at the same time name "localhost" is not accepted? Host mame string is not need for connect to server, need only port. What a  function is hostname doing?

EmBitz, 2019-12-07:
>  Second:
> Why is default host adress - 0.0.0.0, that address is not valid. But at the same time name "localhost" is not accepted? Host mame string is not need for connect to server, need only port. What a function is hostname doing?

Sure it is, it means "bind to any interface".
https://stackoverflow.com/questions/20778771/what-is-the-difference-between-0-0-0-0-127-0-0-1-and-localhost

Extensions: I will look into that one.

EmBitz, 2019-12-08:
User extensions are allowed in the next update.
Thanks!

closed 2019-12-08 by EmBitz

EmBitz, 2019-12-08:
I also added to possibility to add several scripts (e.g. user or platform specific reset strategies).

Like:
-D stm32-auto,myReset.scr 

for the stm32-auto the .script will be used because no extension
for the myReset.scr this file name will be used because it has an extension

seniorandre, 2019-12-08:
In other words. the main script is always without an extension and additional scripts always with the expansion?

EmBitz, 2019-12-08:
No, if you give the main script an extension, it will uses that one.

Every script filename is checked if there is an extension, otherwise it will use .script. It needs an extension.

seniorandre, 2019-12-08:
Ok I understood you. Ican use scriptname with extension and w/o extension.

EmBitz, 2019-12-08:
Yip, and w/o extension will result in default <file>.script