EmBitz/EBlink #11: In-memory debugging

AVI-crak, 2019-12-08:
Hello.
I added one line, now the program can be executed in memory. stm32f746
`	/// SCB->VTOR = 0x20000000;
	result = itrfApi.writeMem32(0xE000ED08, 0x20000000);`

Probably you need to find a way to read project files, where the address of the .isr_vector section is indicated. In order to work with flash memory, or ram memory - separately.

Miinik, 2019-12-08:
This is usually solved in a different way. In file system_stm32xxx (for example system_stm32f4xx.c) there are lines. 
```
  /* Configure the Vector Table location add offset address ------------------*/
#ifdef VECT_TAB_SRAM
  SCB->VTOR = SRAM_BASE | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal SRAM */
#else
  SCB->VTOR = FLASH_BASE | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal FLASH */
#endif
```
You can see them in the screenshot in this comment. https://github.com/EmBitz/EBlink/issues/7#issuecomment-562765102

In the build options on the #defines tab, add VECT_TAB_SRAM.
[image]

AVI-crak, 2019-12-08:
Code written in SystemInit - SCB-> VTOR = 0x20000000; - not executed, the contents of the SCB-> VTOR register are not changed. Above and below - the effect is the same. However, through editing the memory, a new value can be written.
Can debug it overwrite it back?
However, the same operation in the script is performed and is beneficial.

EmBitz, 2019-12-08:
The way Miinik is doing this is preferable. You want to keep the script files as generic as possible.

However, from version 1.1 you can define multiple scripts so you could integrate a user reset strategy on the CLI like:

eblink -I stlink -D stm32-auto,myreset -G

in the file myreset.script you could place your per project user reset.

In the next release ( >1.1) it is even possible to make dynamic hooks in the script like:

 if( isScriptObject("myfunction") )
{
   result =    myfunction(para);
}

Like this you can first check if a certain object (so also functions) exists and if so call it. Like this, the sky is the limit.  The next priority is to document the scripting API.

AVI-crak, 2019-12-08:
EmBitz - you said that the new version of EBlink has an additional program delay when recording and erasing flash memory.
But instead of waiting, you need to poll the status, for this there are registers in the periphery. software delay is needed as a freeze protection - to report a recording failure.

- Your own functions in scripts. This is an interesting direction.

Miinik, 2019-12-08:
>  GDB with the monitor command writeu32.

Good idea. It works.

[image]
[image]

AVI-crak, 2019-12-08:
>  You want to keep the script files as generic as possible.> 
Advanced scripts? - Probably you can say that.
All that can be automated is worthy of a single job. In order to get the "make beautiful" button as a result.
If EBlink is a smart cache between the GDB server and the hardware level, then to create effective scripts, you need a public API. With a description of all available functions, their dependencies, and general algorithms.

EmBitz, 2019-12-09:
@AVI-crak I don't understand your comment about that delay. Do you mean in the script between program cycles? I discovered that working with external loaders vs direct writing that the later is as fast or even faster than with an external loaders. Programming takes a fixed amount of time in the [us] and the interfaces are always too slow  so you can write directly without all that busy polling. That's also how j-link is doing it and how they get those high speed programming figures.

Yes, documentation is crucial. I use EBlink now for almost 4 years for  my personal use only  but I decided some weeks ago to make it more suitable for a broader public  now Texane/stlink is on a dead track. I'm working on a large H7 project and having stlinkV3 makes a huge difference in speed. 

But I don't know if the community is waiting for another tool so I'm observing how it is received (or not :)

AVI-crak, 2019-12-09:
Oh yeah! It is j-link v9.6 that stumbles on the Chinese GD32F. There, the waiting time depends on the memory chip glued on top, always a lottery. And if the code is executed in RAM, it is difficult to make an exact universal delay.
StlinkV3 is cool but expensive. Fifteen times more expensive than the Chinese stlink V2. Which segment has more users?
In addition, users using the stm32hxx and stm32mp1xx series are very, very, few.

Additionally (irrelevant) - how do you manage to operate by placing data in the necessary sectors and blocks through function parameters - without errors ??? I constantly had errors ... Therefore, I wrote a function that simply accepts the address, data and size. https://bitbucket.org/AVI-crak/system_f746/src/default/qspi_driver_S25FL256S/qspi_rw.c
Now it’s enough to set three parameters, and do not care about physics.

EmBitz, 2019-12-09:
>  how do you manage to operate by placing data in the necessary sectors and blocks through function parameters - without errors ???

I'm parsing the XML memory map to build a sector table for  e.g. caching. A sector has, beside the data, also the start address, length and sector number. I'm starting normally with number 0 and increment on every new sector, however with the XML attribute "secstart" I can change the number from which the sector counter will start from that point. You could look at the stm32h7x.script, I introduce a fake sector there with #999 for holding the option bytes. This fake sector is filtered at the flash erase and write to write the option bytes.

closed 2019-12-09 by EmBitz