EmBitz/EBlink #12: Slow debugging in STM32F4

Miinik, 2019-12-10:
With step-by-step execution in microcontrollers STM32F4 (tested on F401, F407, F411) steps are slow (more than a second / step).
In STM32F0 and STM32F1, there is no this problem.

EmBitz, 2019-12-11:
Could you use -v7 on the cli to check what is going on?

Miinik, 2019-12-11:
This is the same project. But I tested other projects with STM32F4 and a similar problem.

One step with argument -v7.
[image]

I noticed that the data in the EBlink console appears slower when debugging STM32F4 compared with debugging STM32F1.

EmBitz, 2019-12-11:
I just tested a F429, lightning fast! 
Also my H753 project is very fast but that one is using stlinkV3.

I think that this has something to do with the speed of the on-board st-link. I think that you don't see the "Interface speed: xxxx" notify line at startup. That means that the on-board stlink doesn't support variable speed and the speed is fixed at 1800Khz (If 'm right). 

Normally EBlink will set the maximum speed of the interface if there is not an explicit speed on the CLI. So, if you see the "Interface speed: 4000Khz" then you know that the ST-link supports multiple speeds and that it is at the max for a stlinkV2 (v3 max = 24000Khz).

Miinik, 2019-12-11:
EBlink reports that "Interface speed: 4000KHz".

EmBitz, 2019-12-11:
No, I can't reproduce it. My F407VG Nucleo is a bit slower but that's because it is a fixed 1800Khz on-board stlink (+/- 600ms per step). All the 4000KHz and higher are more or less, instantly.

Also, the debug code is the same for all the cortex-m cores, that's one of the nice things about ARM.

EmBitz, 2019-12-12:
I close it for now, if we know more we will reopen it.

closed 2019-12-12 by EmBitz

Miinik, 2020-08-07:
As it turned out, slow debugging on ALL microcontrollers.
For example, debugging on STM32F103.
This is an animation of debugging using EBlink.

[image]

And this is an animation of debugging using STLinkGDB.

[image]

STLinkGDB has many times faster debugging speed.

EmBitz, 2020-08-10:
Hmmm, strange. we are using EBlink heavily daily and we don't notice any problems. I'm using a 3900X amd windows 10 machine. Perhaps the difference in libusb (I'm using a newer release for EBlink ) is making a difference on your machine.
Nevertheless I will do some tests today to see is I can find some differences in timing with a USB sniffer.

Miinik, 2020-08-10:
I used a USB sniffer. One step in the code, requires 2512 USB transactions and takes 1.2 seconds.

[EBLink USB sniffer.zip](https://github.com/EmBitz/EBlink/files/5049733/EBLink.USB.sniffer.zip)

EmBitz, 2020-08-10:
I made a screen capture of a H7 debugging (out current project). EBlink is running on a I5 remote at a workbench so there is also an Ethernet connection in the middle between IDE and GDB server.

https://www.embitz.org/eblink.wmv

Miinik, 2020-08-10:
I used ST-Link v2, which has a USB speed of 12 mb/s and SWD at 4 MHz.
ST-Link v3 has a USB speed of 480 mb/s and a SWD speed of 24 MHz.
Perhaps due to the higher speed of ST-Link v3, you do not notice a slowdown in debugging.

EmBitz, 2020-08-10:
The problem is that I can't test it anymore. All those composite USB STlink's are not working anymore on windows 10 and STlinkGDB. I only can use EBlink. Otherwise I could compare them side-by-side with a sniffer and see where the lag is (if any).

EmBitz, 2020-08-10:
Ok, I have switched the V3 to stlink V2 and to be honest, I don't notice any debug difference. Only the image flashing takes much longer.

https://www.embitz.org/eblink2.wmv

EmBitz, 2020-08-10:
How is ctrl-F10 (instruction step) working? If I use ctrl-F10 with stlink V2 and EBlink it is more or less instantly.

Miinik, 2020-08-10:
The your video shows a slight slowdown in debugging speed starting from the function **scheduleEvent**.

Ctrl+F10 works very fast. But F10 or F11 is slow.

I have RaspPi, but I have no idea how to debug without an IDE...

I noticed that the simpler the microcontroller, the faster the debugging. On F0 and F1 the debugging speed is acceptable, but less than that of STLinkGDB. On F4, F7 and H7, the debugging speed is slow.
If the line contains multiple logical lines separated by semicolons. This greatly reduces the speed of debugging. https://github.com/EmBitz/EBlink/issues/12#issuecomment-670425182

I used several ST-Link with different firmware versions (2.28.18, 2.32.22, 2.34.25, 2.36.26) and it did not affect the work and the speed of debugging.

EmBitz, 2020-08-10:
Yes, because those events are critical and a lot of logic is attached at those events.

I think that what you are seeing is that GDB is using two different ways of single source step. GDB can either use a loop of stepi (instruction steps) to walk to the next source line or it can use a set/remove breakpoint method. The later costs a resource but is faster.  I don't know how GDB decides which one to use but perhaps STlinkGDB is forcing GDB to use the breakpoint way.

Your test code is also slow on our board because it is single stepping through all the machine instructions. You can watch this by verbose level 7 of EBlink.

Can you watch (wire shark perhaps) if STlinkGDB is driven by GDB with breakpoints?  Syntax z<n>,<addr>

Would be interesting to figure out how this switching between both single step mechanisms is implemented.

Miinik, 2020-08-10:
> Can you watch (wire shark perhaps) if STlinkGDB is driven by GDB with breakpoints?

I do not know how to do that.

EmBitz, 2020-08-10:
I'm flabbergasted, it has something to do with GDB but I don't know why. I'm already digging in the GDB code to check how that decision is done. If I have more time I will try to get STlinkGDB back working and run GDB in my debugger.

EmBitz, 2020-08-11:
Well, it is not a EBlink or STlinkGDB issue but it is how GDB determines how to do the single source step. That explains that the more simple MCU's are going faster because GDB is using the breakpoint methode there. It has something to do with automic instructions and you can find the implementation in GDB arm-tdep.c.

A simple test shows this, I rewrote your test program and as you can see, the first 9 lines are very fast.

https://www.embitz.org/eblink3.wmv

I will see if a newer GDB version will tackle this.

Miinik, 2020-08-11:
Thanks.

I have tried replacing GDB  from different versions the distribution of GCC arm-none-eabi.
Debugging starts but doesn't work. Possibly in EmBitz, modified version GDB.

EmBitz, 2020-08-11:
Yes, EB is using special tailored GDB because of e.g. live variables but also some missing commands are added.

I have already downloaded GDB 8.3.0 which I will modify to EB needs.

EmBitz, 2020-08-12:
But maybe we can also call this "bad coding style" for GDB debugging. It's better to avoid multiple statements on one source line, as it turns out.  And to be honest, it makes sense. If I recall correctly, Keil also had speed problems with them, at least in the past.

But it would be nice that there was a GDB switch to force breakpoint use (if any breakpoints left because that's the down side) and I will see if I can add this. At the moment I'm very busy.