EmBitz/EBlink #13: EBlink integration EmBitz 1.11.

EmBitz, 2019-12-18:
I added EBlink to list of debugger interfaces for EmBitz 1.11.

[image]

To add, you need to extract this archive https://www.dropbox.com/s/4xzul9eagrpakju/EBlink_Interface_for_EmBitz_1_11.zip
to a folder 

> ${EMBITZ}\share\EmBitz\debuggers\Interfaces\  

EBlink must be installed in the folder

> ${EMBITZ}\share\EBlink

**${EMBITZ}** this is the folder in which EmBitz is installed. Usually this 

> C:\Program Files (x86)\EmBitz\1.11\

_Originally posted by @Miinik in https://github.com/EmBitz/EBlink/issues/6#issuecomment-564215742_

Miinik, 2020-01-21:
Updated debug interface for [EBlink 2.0](https://github.com/EmBitz/EBlink/releases/tag/2.0)
https://www.dropbox.com/s/ylyhi1b826b5fq7/EBlink_2.0_Interface_for_EmBitz_1_11.zip

Miinik, 2020-10-29:
Fixed display of window contents at DPI scale greater than 100%.

[EBlink_Interface.zip](https://github.com/EmBitz/EBlink/files/5461338/EBlink_Interface.zip)

EmBitz, 2020-11-05:
Also, update the config.script file

[config.zip](https://github.com/EmBitz/EBlink/files/5496034/config.zip)