EmBitz/EBlink #14: Need command for disables cursor animation

Miinik, 2019-12-18:
Need command for disables cursor animation, because cursor animation creates garbage in stdout.
Here is the current result from stdout.

> tool 'eblink': C:\Program Files (x86)\EmBitz\1.11\share\eblink\eblink.exe -I stlink,swd -D stm32-auto -F erase,verify,run,file="D:\STM32\STM32F103\Test\bin\Release\Test.hex" (in C:\Program Files (x86)\EmBitz\1.11\share\eblink)
> stdout> 
> stdout> EBlink version 1.1-[18] by Gerard Zagema
> stdout> 
> stdout> Interface at  1:1 - <BUS>:<ADDR>
> stdout> Interface type : STlink/V2
> stdout> STlink connect : Under reset
> stdout> Target voltage : 3.22V
> stdout> Interface speed: 4000KHz
> stdout> Target detected: Cortex-M3 (r1p1) 
> stdout> HW breakpoints : 6
> stdout> HW watchpoints : 4
> stdout> Fault unwind   : Enabled
> stdout> STmicro device : 0x410
> stdout> Detected FLASH : 0x20000
> stdout> Configured RAM : 0x5000
> stdout> Reset: system
> stdout> 
> stdout> HEX file: D:\STM32\STM32F103\Test\bin\Release\Test.hex
> stdout> 
> stdout> /
> -
> \
> |
> /
> -
> \
> |
> /
> -
> \
> |
> /
> -
> \
> |
> /
> -
> \
> |
> /
> -
> \
> |
> /
> -
> \
> |
> /
> -
> \
> |
> /
> -
> \
> |
> /
> -
> \
> |
> /
> -
> \
> |
> /
> -
> \
> |
> /
>  
> stdout> Flashing sector  0, address: 0x08000000, length: 0x0308
> stdout> Flash verify .. done
> stdout> Flasher starts target
> stdout> Reset: system
> Tool execution terminated with status 0

EmBitz, 2019-12-18:
Hi, yes I see.

The current verbose map is as below.  Perhaps you can find a suitable mapping, ( -v0 if first etc..)
I will add an additional switch for the next version.

```
/// @brief Verbose level mapping, we could define special combinations if needed
const unsigned int verboseMap[] =
{
    HDR_INFO|ERROR_INFO, // 0
    HDR_INFO|ERROR_INFO|USER_INFO|ANIMT_INFO,
    HDR_INFO|ERROR_INFO|USER_INFO|SCRIPT_INFO|ANIMT_INFO,
    HDR_INFO|ERROR_INFO|USER_INFO|SCRIPT_INFO|TARGET_INFO|ANIMT_INFO,
    HDR_INFO|ERROR_INFO|USER_INFO|SCRIPT_INFO|TARGET_INFO|NOTICE_INFO|ANIMT_INFO,
    HDR_INFO|ERROR_INFO|USER_INFO|SCRIPT_INFO|TARGET_INFO|NOTICE_INFO|WARN_INFO|ANIMT_INFO,
    HDR_INFO|ERROR_INFO|USER_INFO|SCRIPT_INFO|TARGET_INFO|NOTICE_INFO|WARN_INFO|DBG_INFO,
    HDR_INFO|ERROR_INFO|USER_INFO|SCRIPT_INFO|TARGET_INFO|NOTICE_INFO|WARN_INFO|DBG_INFO|TRACE_INFO, //7
};
```

EmBitz, 2019-12-18:
I think that I will introduce verbose levels 10 to 17 which is equal as 0 to 7 but with animation suppressed. BTW default verbose level is 4 so with -v 14 you will get the default behaviour without animation.

EmBitz, 2019-12-18:
The new switch will be -a for no animation (check help for more options)

[eblink.zip](https://github.com/EmBitz/EBlink/files/3980011/eblink.zip)

EmBitz, 2019-12-18:
Tip: you could skip the erase if you don't mind that unused sectors are not cleared. That way only modified sectors will be (re)written.

closed 2019-12-19 by EmBitz

EmBitz, 2019-12-19:
Hi,

I changed my mind about the -a, it will be -V (capital v) with an optional verbose level in the next release.
So you have to change -a in -V and you could use an optional verbose level like -V 5. 
The -v needs still an argument otherwise it would be useless.

```
  -h,           --help                  Print this help
  -g,           --nogui                 No GUI message boxes
  -v <level>,   --verbose <0..7>        Specify level of verbose logging (default 4)
  -V [level],   --noanimation [0..7]    Same as -v but without cursor animation
  -D <scripts>, --device <scripts>      Device script load list <.,.,.>
  -T <options>, --target                Select target(optional)
  -I <options>, --interf                Select interface
  -F <options>, --flash <options>       Run image flashing
  -G [options], --gdb <options>         Launch GDB server
```

Miinik, 2019-12-19:
This will lead to a conflict of commands. 
Suppose on the command line there are arguments.

> -v2 -V4

Which command has high priority?
Using **-a** does not have this problem.

EmBitz, 2019-12-20:
I think I will do it differently. Sometimes, like full chip erase, an action can take more than 30 seconds or longer. So even in non-animation mode it would be nice to have some feedback.

So I'm thinking of introducing the -p [progress type]

-p 0 = non progress at all
-p 1 = animated cursor             
-p 2 = fast dot, every 500ms     
-p 3 = normal dot, every 1 sec
-p 4 = slow dot, every e.g. 4 sec

Default will be 1 and -p without any optional type will default to -p 0. 

Or (type > 1) will give dot progress with (type-1)*500ms so -p 6 will give 2.5 seconds between dots

The dot progress is just '.' without any special terminal escape sequences.

EmBitz, 2020-01-06:
Update:
This is implemented as the -a  command line switch