EmBitz/EBlink #15: EBmonitor not working?

dalbert2, 2019-12-28:
Thanks for posting EBlink (and instructions for use with EmBitz 1.11)!  It's allowed me to start developing on the STM32G0 platform.

Unfortunately, I haven't been able to get EBmonitor working when using EBlink; is it expected to be working?  The plugin is enabled, the buffers are created and registered, but when printf() runs, the program immediately ends up in the default exception handler.  I'm probably doing something stupid, but just want to make sure that it is expected to be working.

AVI-crak, 2019-12-28:
printf () needs to be configured, you must specify the print direction.
I use my bike with square wheels. It allows you to write code without thinking about the consequences.
https://bitbucket.org/AVI-crak/sprint/src/default/sPrint.c
[image]

Miinik, 2019-12-28:
You also need to enable EBmonitor in the project options.

[image]

With STM32F103 and EBlink, EBmonitor works.

EmBitz, 2019-12-28:
Yip, EBmonitor should work with any debugger (GDB + GDB server combi) which supports live variables. There are no special EBmonitor specific features inside EBlink.

closed 2019-12-28 by EmBitz

dalbert2, 2019-12-28:
Thanks folks.  I have used EBmonitor before and did have the plugin enabled and the linker flag set.  It has been a while, but I don't recall having to implement a softPrint function; IIRC, when the linker flag was set, an appropriate module was linked in that redirected printf to GDB.

I'm using an STM32G0 target (which isn't directly supported by EmBitz but works fine when using EBlink and appropriate header/startup files).  I'll see if I can reproduce the problem with an STM32F103 target to see if the issue is related to the target.

EmBitz, 2019-12-28:
Ah, ok. Check the map file and see if the ebmonitor related stuff is pulled in.

dalbert2, 2019-12-28:
Good idea!
libebmon.a is being pulled in and _eb_monitor_stdin and _stdout are defined in data along with their buffers.  Also __sf_fake_stdin, _stdout are defined in ro_data.

I am pretty sure it's something I've done wrong in configuring the linker for the STM32G0.  What's happening is a null pointer de-reference in puts() because the string constant isn't being processed correctly...I'll figure it out from here, thanks for the help!

dalbert2, 2019-12-28:
Yep, I messed up the linker script...it's working properly now, thanks again!

gopalreddylakki, 2020-11-07:
its working