EmBitz/EBlink #16: EBmonitor plugin

AVI-crak, 2020-01-16:
Good afternoon.
There is a desire to take part in improving the EBmonitor plugin.
The fact is that its optimization does not take into account the capabilities of the communication interface. There is optimization in terms of quantity and speed of code - which runs on both sides of the interface. But the communication interface is too slow, and its limitations must be considered.
Using a smart data cache does not save the situation. Because the smart cache takes into account the properties of data types, if the data is marked as changing outside of visibility (const volatile), reading will be unconditional.
To reduce confusion, I will give my own version of the structure, which almost coincides with yours.
static struct _stdout
{
    const uint16_t length;
    const volatile uint16_t tail;
    volatile uint16_t head;
    uint16_t mode;
    volatile char * ptr;
} _eb_monitor_stdout = {eb_buf_zize_out, 0,0,0, & _ std_out_buffer [0]};

The fields of the structure that can and should be read only once: length, tail, * ptr, mode.
Head counter - read once per one cycle of time allotted for the EBmonitor plugin to work. I know that the EBmonitor plugin is not the only one that uses the communication interface. To ensure operational data control, requests to the communication interface are forcibly limited at a certain level - so that all plugins can receive their small portion of data.

This is exactly what is the first point of acceleration - do not read the head counter too often. You read it once - and you already understand how many data bytes you need to read.
Reading one byte is very expensive. To do this, each time you need to transmit the full address, mask, and read command.
Read all 32 bits, without a mask, with automatic address increment. This will reduce data processing time.
I’m not sure if there is a problem, but if there is a difficulty in converting the data presentation format, it is much more profitable to process on the IDE side. There is a powerful processor, and a lot of memory - let them work.

The tail counter is read only once, after which it is no longer necessary to read it - you need to write it. It also makes no sense to write too often. Only at the moment when there are differences between the old record in the new * ptr read cycle.

That is, the general algorithm (for one plug-in cycle) should look like this:
1 Update tail - in case there were changes in its state in the previous cycle.
2 We read head once !! We consider how much data we need to read, we calculate the amount of data that we can read in this cycle.
3 Read the data from * ptr into a virtual buffer (exactly the same as on MK). We read at high speed, without dividing into individual bytes. Debriefing after the appearance of data in a virtual buffer.

The EBmonitor plugin is not a real-time device. No need to program it for an extremely fast response. It is closer to asynchronous communication types - where the response may have an uncontrolled delay, but it is always relevant.

Semantic errors - the problem of translating text from Russian. I tried to reduce their number.

EmBitz, 2020-01-16:
Hi,

This is not the right channel to discus Embitz, but ok.

The GDB debugger plugin has a live event ticker, if the connected GDB server supports live variables then all the GDB debugger plugin-plugins are catching that live ticker (e.g. also the OS viewer) event to get information. And on every live tick event the EBmonitor plugin is sampling the pointers of the circular buffers to see if there are any characters to get.  It needs to parse the MI formatted answer because the answer are coming from GDB.exe.  It is calculating the amount of data in the circular buffer and it is not asking it byte for byte.

void EBmon::OnGdbAnswerStdout( EmBitzEventByName& event)
{
    unsigned long tNum;

    // Check if we have a valid answer
    if(event.GetInt() == 0)
    {
        //print error in the log pane

        m_pEbmonLog->Clear();
        m_pEbmonLog->Append(_T("The EBmon.lib is not found."), (Logger::level)(Logger::error));

        m_busy=false;
        m_invalidTarget = true;
        return;
    }

    // This variable is misused as a original query storage
    // wxString queryStr = event.GetOldBuildTargetName();
    wxString valueStr = event.GetString();

    valueStr = valueStr.AfterFirst('{');
    valueStr = valueStr.BeforeLast('}');

    miParserClass miParser(valueStr);

    m_length = -1;
    m_tail = -1;
    m_head = -1;

    miParser.GetFirstKey(_T("length"),true,true).ToULong(&m_length);
    miParser.GetFirstKey(_T("tail"),true,true).ToULong(&m_tail);
    miParser.GetFirstKey(_T("head"),true,true).ToULong(&m_head);

    if( (m_length == -1) ||
        (m_tail   == -1) ||
        (m_head   == -1) )
        {
            m_pEbmonLog->Clear();
            m_pEbmonLog->Append(_T("The EBmon.lib is corrupted."), (Logger::level)(Logger::error));
            m_busy=false;
            m_invalidTarget = true;
            return;
        }

    // Check if we have something in the buffer
    if( m_tail == m_head)
    {
        m_busy=false;
        return;
    }

    if(++m_tail >= m_length)
        m_tail = 0;

    wxString queryStr = wxString::Format(_T("-data-evaluate-expression \"_eb_monitor_stdout.ptr[0]@%d\" 0 d"), m_length);
    EmBitzEventByName _event(_T("EVT_DBG_RAW_QUERY"));
    _event.SetString(queryStr);
    _event.SetId(idEBmonOutBuf);
    Manager::Get()->ProcessEvent(_event);
}

EmBitz, 2020-01-16:
I closed this issue but you can still response.

closed 2020-01-16 by EmBitz

EmBitz, 2020-01-17:
P.s. Reading the head or tail once doesn't make any noticeable different in speed. The EBmonitor is asking GDB for the whole struct so all the infoamtion is comming with just 1 query and not separate queries.  There is only 1 address transfer and than the whole block is transferred. This is faster than individual fields. When I developed EBmonitor I did several tests with different strategies. The fastest is getting the whole struct and circular buffer instead of all kind of separate GDB queries. A query is more  expansive  than a data block.

AVI-crak, 2020-01-17:
I used to print one character very quickly (test), as a result I got a very dense status log _eb_monitor_stdin.tail, when the clipboard was full (printing is faster than the interface exchange speed). This result gave a misconception about the logic of the plugin.

I wrote another simple test - committing changes to _eb_monitor_stdin.tail with saving time in a small buffer buffer.
The results confirm your words. The counter _eb_monitor_stdin.head is read once per full cycle.
However, the cycle itself is completely executed - until its logical conclusion, in one large piece of code.
Problem - the cycle runs completely, without dividing by time slices. When the clipboard is fast and full, it is blocked for the duration of reading from the side of the plugin.
This is not a critical error, but it does cause some problems. The task that prints messages is forced to wait too long.
I understand that the task is in any case forced to wait, because the problem is connection speed. However, the overhead of the operating system — switching with saving and restoring the context — is spent on the task itself. This time is wasted if the task has nothing to do.

To exclude lengthy full locks - the counter status _eb_monitor_stdin.tail should be updated with a fixed minimum time (say 100ms), and at the end of the buffer reading cycle.

I have one more suspicion that I can’t test with tests, but I saw a similar problem in other people's terminal programs. Perhaps a delay appears while printing text in the IDE itself. For example, the popular terminal program Terminal 1.9b - suffers from a similar problem. There, the communication interface can receive data at a crazy speed, there are no restrictions. However, printing itself will be very slow and long.

This problem is not a priority, and is more like a final polishing of furniture. If something is burning on you, then the appearance is no longer important.

EmBitz, 2020-01-17:
Are you aware of the internal buffering of the standard pipes at the GNU libraries?
You can set thresholds and you can use flush commands.
Do you have the source files of the ebmon target library?

EmBitz, 2020-01-17:
[ebmon.zip](https://github.com/EmBitz/EBlink/files/4077168/ebmon.zip)

AVI-crak, 2020-01-17:
If I understand correctly, ebmon.zip is software for hosting on the MK side. For me, these functions look a little different, since they work as part of the OS. It can print several tasks at once, at different times. To eliminate errors, additional data processing is performed.
/// print text
/// sTask_skip() - Command for OS, task forced switching
/// sSystem_task.system_us - System time, step 1ms
void monitor_print (char* text)
{
    if ( (text[0] == '\0') || (_eb_monitor_stdout.mode == 2) ) return;
    while (_eb_monitor_stdout.mode == 1)sTask_skip();
    _eb_monitor_stdout.mode = 1;
    char* in_txt;
    uint16_t temp_h;
    int32_t temp_tim;
    /// two seconds to freeze the interface
    temp_tim = 0 - (2000 + sSystem_task.system_us);
    temp_h = _eb_monitor_stdout.head + 1;
    in_txt = text;
    do
    {
        if (temp_h == eb_buf_zize_out) temp_h = 0;
        if (temp_h == _eb_monitor_stdout.tail)
        {
            if ((sSystem_task.system_us + temp_tim) > 0)
            {
                _eb_monitor_stdout.mode = 2; return;
            }else sTask_skip();
        }else
        {
            _eb_monitor_stdout.ptr[temp_h] = *in_txt++;
            _eb_monitor_stdout.head = temp_h;
            temp_h++; __DSB();
        };
    }while (*in_txt != '\0');
    _eb_monitor_stdout.mode = 0;
};

The GNU structure and general algorithm of operation - I vaguely imagine. EBmonitor plug-in software - not seen. For me it is more like a black box in a dark room. However, there are ideas for improving EBmonitor.
You just need to add a time counter to the internal clipboard reading cycle. And already according to his condition, update _eb_monitor_stdin.tail - for the case when he managed to change.

AVI-crak, 2020-01-30:
Thanks for posting the IDE.
I was looking for a trace of the EBmonitor plugin in the IDE, there are very few references. There is a certificate of connection and registration, but there is no code and its assembly.
Also, the functions and methods that are your message "void EBmon :: OnGdbAnswerStdout (EmBitzEventByName & event)" do not exist there. This is the one that interests me very much.

EmBitz, 2020-02-05:
[ebmon.zip](https://github.com/EmBitz/EBlink/files/4157921/ebmon.zip)

AVI-crak, 2020-02-08:
Hi EmBitz.
I found a problem with the debug module, which is located in the Cortex core. The debug module performs memory modification operations (read> change> write) - along the edges of the external interface (SWD, JTAG).
The external interface is much slower than the Cortex core frequency. As a result, writing a single byte is not atomic.
Simple code managed to catch transients during memory modification.

I check _eb_monitor_stdout.tail for changes every 500ns. As a result, there is a stable print job at a high frequency of the Cortex core, with a large clipboard.
`__attribute__( ( always_inline ) ) static inline void delay( uint32_t volatile time_tmp){do {time_tmp--;} while ( time_tmp );};
/// print text
/// os_pass() - Command for OS, task forced switching
/// os_data.system_us - System time, step 1ms
void monitor_print (char* text)
{
    static uint32_t status = 0;
    if ( (text[0] == '\0') || (status == 2) ) return;
    while (status == 1)os_pass(); /// busy there
    status = 1;
    char* in_txt;
    uint32_t temp_h, temp_t;
    int32_t temp_tim;
    temp_tim = -2000;
    temp_tim -= os_data.system_us;
    temp_h = _eb_monitor_stdout.head + 1;
    in_txt = text;
    do
    {
        do {temp_t = _eb_monitor_stdout.tail; delay(50);}
        while (temp_t != _eb_monitor_stdout.tail);
        if (temp_h == eb_buf_zize_out) temp_h = 0;
        if (temp_h == temp_t)
        {
            if ((os_data.system_us + temp_tim) > 1)
            {
                status = 2; return;
            }else os_pass(); /// busy there
        }else
        {
            _eb_monitor_stdout.ptr[temp_h] = *in_txt++;
            _eb_monitor_stdout.head = temp_h;
            temp_h++; __DSB();
        };
    }while (*in_txt != '\0');
    status = 0;
};`

EmBitz, 2020-02-21:
Hi AVI-crak ,

Thanks! could you share the files?

AVI-crak, 2020-02-21:
The discussion subject does not have program code - it is a hardware implementation of the debugger module, which in turn is part of the ARM core.
The fact is that the debugger module was designed before the first commercial version of ARM. At that time, everything, or almost everything, used a direct data storage order. This left its mark on the hardware implementation of the debugger module. A structure with bit fields in direct byte storage mode is a mess, which is very difficult to understand. The reverse order of bytes and bits turned out to be surprisingly successful for ARM processors. It allowed to save a lot of silicon, and significantly speed up the algorithms.
For example, changing the order for four bytes turned out to be quite simple for a hardware implementation.
But everything else required preliminary reading and writing by one byte. Engineers have always saved silicon, but in this situation nothing can be changed - the backward compatibility of generations is violated.

The first ARM processors did not know how to hardware-wipe an insignificant part of the data when reading one byte; for this, they introduced a separate program command. More modern processors - hardware cleans up an insignificant part of the data when reading / writing one byte. But even in this case, backward compatibility was preserved in GCC, which even in the latest version uses a useless cleaning command.

Backward compatibility imposes restrictions in further development, but nevertheless there is a way out.
Minimize the use of writing / reading data less than the size of the register (4 bytes). In this case, there is a noticeable acceleration.

A suggestion for the EBmonitor debugging software module is to use a union for an existing structure, with access in uint32_t * mode. Part of the data with the structure does not change, this allows you to rewrite the constant part without distortion (field length).

EmBitz, I apologize for my level of knowledge. I was not able to build an IDE build environment from your sources. I fought for a very long time, but the pop-up build errors killed my patience.

EmBitz, 2020-02-22:
AVI, as soon as I have time I will help you.

aroyer-qc, 2020-07-24:
Just to let know to people that want this to work and you don't understand why it doesn't work... 
If you have created your project by yourself... you need to put a call at the beginning of your code (if you did not use EmBitz wizard - CRT0)

the call is 
software_init_hook ();
This initialize the buffer for EBmonitor and then it should work..