EmBitz/EBlink #17: Memory size

Miinik, 2020-01-17:
In many STM32 microcontrollers amount of memory (flash, RAM) is more than specified in the datasheet.
For example, in F103C8, the datasheet is 64 KB, but actually 128 KB.
In H750VB in datasheet is 128 KB, but actually 2048 KB flash memory.
In EBlink scripts, size of flash memory is read from F_SIZE register (address 0x1FF1E880) and not always equal to the actual size of flash memory.
A command is needed that will override the size of the flash.

> -M FlashSize=2048

That is, if this command is not present, the script reads the size of the flash from the F_SIZE register (as it is now).
If there is a command, the specified size is used.

```
     // Get the flash size by reading the info register
    if (FlashSize<=0)
    {
       result = itrfApi.readMem32(0x1ff1e880);
       flash_size = (itrfApi.value32 & 0xffff) * 1024;
    }
    else
    {
       flash_size = FlashSize * 1024; // Flash size specified by the -M command.
    }
```

Now as a workaround, I edited the script stm32h7x.script specifying explicitly the size of 2048 KB.
```
    // Get the flash size by reading the info register
    //result = itrfApi.readMem32(0x1ff1e880);
    flash_size = 2048 * 1024;   //(itrfApi.value32 & 0xffff) * 1024;
```
This works and loads firmware in the H750VB, but it is not a universal solution.

EmBitz, 2020-01-17:
There is a workaround currently.
Define FlashSize as global in a separate script file e.g.  myFlashSize.script

Call EBlink with: 

` -D stm32-auto,myFlashSize`

I think that that should work

Miinik, 2020-01-17:
This is not a universal method. Suppose I put the project on the Internet. Other people do not have myFlashSize.script and this method will not work.

EmBitz, 2020-01-17:
I agree, that's why I called it a workaround and not a solution.

I'm wondering how STutil from STmicro is getting the right information about the sizes? If STutil can differentiate the MCU's then it should be possible without commandline hacks. 

But I agree that this would be a useful addition and I put it on my list. However, I want to keep the cli as simple (clean) as possible. I don't want to build a second openOCD. So if we can solve this the STutil way that would be nice.

Miinik, 2020-01-17:
I noticed that ST-LINK Utility is ignoring size information. Just reads and writes the given addresses. If this is not possible, it reports an error.

Miinik, 2020-01-17:
Now the scripts are in the folder with EBlink.
It would be great if there was an opportunity to place the user script in the project folder.
To do this, somehow inform EBlink the path to the project folder so that it loads the script from there.
In this case idea with myFlashSize.script will work.

EmBitz, 2020-01-17:
Isn't STutil showing the memory sizes in the right upper corner?

Miinik, 2020-01-17:
Shows flash size, but does not use when reading / writing firmware.
For example, for STM32F103C8T6, it shows 64 KB, but it allows to flash 128 KB. If the firmware is larger, it flashes 128 KB and then reports an error. That is, flashing as much as possible.
For the STM32H750VBT6, ST Utility showing 2 MB, but in the register F_SIZE the value is 128.
Perhaps all STM32H7 models have a 2 MB flash.

EmBitz, 2020-01-20:
Ok, I made a quick test case.

Because this tool has changed during the initial setup 3 years ago, the switch names are not that obvious anymore. E.g. -D was in the past a device specification and now is it a script file def. So I changed the switch names, better now than if it is used wide spread and change it.

The -S is now script file and you can have as many -S switches as you like.
The -D is script define in format variable=value e.g. flashSize=5 or name=\Miinik\". The typing is automated by squirrel. You may have as many -D as you like.

To use a conditional global in script and prevent errors if that define is omitted, first check if that define exists like:

```
    if( isScriptObject("flashSize") )
    {
        printf("flashSize : 0x%X\n", flashSize);
    }
```

[eblink.zip](https://github.com/EmBitz/EBlink/files/4085671/eblink.zip)

Miinik, 2020-01-20:
Thanks
I added variable FLASH_SIZE to scripts. https://www.dropbox.com/s/5mehzpj077u9rg7/EBlink_scripts_add_FLASH_SIZE.zip

For flash size 2 MB `-D FLASH_SIZE=2048`

Miinik, 2020-01-20:
Updated archive https://www.dropbox.com/s/9v9biqgd5uwg6rb/EBlink_scripts_add_FLASH_SIZE.zip

In file **stm32l1x.script**, I found strange code.
```
        case 0x437 : // CHIPID_STM32_L152_RE
            flash_size = flash_size * 1024;
            page_size = 0x100;
            ram_size = 0x14000;
            break;
```
Replaced by
```
        case 0x437 : // CHIPID_STM32_L152_RE
            flash_size = 512 * 1024;
            page_size = 0x100;
            ram_size = 80 * 1024;
            break;
```

closed 2020-01-20 by Miinik

EmBitz, 2020-01-22:
Added the scripts to the repro  (I added notification if size is set explicitly)