EmBitz/EBlink #18: Script error: Unable to open binary file

Miinik, 2020-01-22:
In EBlink 2.0 an error occurs during the firmware STM32F1.

[image]

If in file **stm32HW_fl.script** replace 
`    dataStart = itrfApi.loadBin("scripts/stmicro/flash/stm32HW_fl.bin", 0x20000000);`
on
`    dataStart = itrfApi.loadBin("stmicro/flash/stm32HW_fl.bin", 0x20000000);`
then there is no error.

closed 2020-01-22 by EmBitz

Miinik, 2020-01-22:
In file **stm32L4_fl.script** also need to be replaced
`    dataStart = itrfApi.loadBin("scripts/stmicro/flash/stm32L4_fl.bin", 0x20000000);`
on
`    dataStart = itrfApi.loadBin("stmicro/flash/stm32L4_fl.bin", 0x20000000);`

In file **stm32-auto.script** in area
```
    // Cortex M3
    case 0x411 : // CHIPID_STM32_F2
        
        // Fix chip_id for F4 rev A errata , Read CPU ID, as CoreID is the same for F2/F4
        result = itrfApi.readMem32(0xE000ED00);
        if ((itrfApi.value32  & 0xfff0) == 0xc240)
        {
            deviceId = 0x413;
            require("stmicro/stm32f4x.script");
            break;
        }
       
        require("stm32f2x.script");
        break;

```
Need to replace
`        require("stm32f2x.script");`
on
`        require("stmicro/stm32f2x.script");`

Miinik, 2020-05-14:
Again, this error.
Need to do as it was before, i.e add in path, folder **scripts**
Very strange...

EmBitz, 2020-05-15:
Sorry, the flash loader couldn't find the image files so I made some modifications.
I didn't tested this. I will look into it.

EmBitz, 2020-05-15:
Ok, this is the problem.

If I look for script files then I try all kind of paths, the ./script and the paths defined on the command line. To load the script I change the pwd and after loading I restore this path. Previous I didn't do that so the pwd was pointing to the latest found script file.

However, using the flash tool to flash a file, the tool was looking in the script directory for that file but the user was expecting the location of the executable EBlink as starting point. So I introduced a restore of the pwd to the executable location.  Problem is that the binary load function is at execution time of the script and the include at loading time.

I changed the loadBinary which is now also using the same search algorithm as the scripts.

Miinik, 2020-05-15:
Thank you the problem is fixed.