EmBitz/EBlink #20: EBlink with embedded STLink v3 from Nucleo board with STM32H745

Polarisru, 2020-04-09:
Hello, I try to use EBlink to communicate with my new Nucleo board with STM32H745 MCU. This Nucleo board has embedded STLink v3, so it should be possible to use EBlink to program it. I have downloaded Win32 version of EBlink and tried it with standard stm32gdb.bat file but without success.
Here is the output of EBlink:
```
eblink.exe -I stlink,dr,speed=3000 -S stm32-auto -G

EBlink version 2.4-[91] by Gerard Zagema

Interface at  1:2 - <BUS>:<ADDR>
Interface type : STlink/V3
Error initializing interface stlink
```
What could be a reason for such behavior?

EmBitz, 2020-04-10:
I have only a Nucleo H743 to test it and I also use it daily with our own board with a stlinkV3 and stm32H753.

If you plugin the board, do you get a popup of file explorer in windows to the emulated memory stick of the stlink? I think it's a driver issue.

EmBitz, 2020-04-11:
You tried the hotplug bat file (with the dr option), also try the stm32gdb.bat

Polarisru, 2020-04-11:
Sorry for the delay, it was not a problem of EBlink, the Nucleo board has some problems with its internal power supply because of two cores probably (H743 has only one Cortex-M7 core and H745 has additional Cortex-M4 core, so the power system is more advanced here), I have fixed the problem and now EBlink can flash the MCU and even debug it. It is really amazing! Thank you for your work!

closed 2020-04-15 by EmBitz