EmBitz/EBlink #21: EmBitz website

dalbert2, 2020-04-23:
I am a long time (paid) EmBitz user and a big fan.  Thank you for the great software Gerard!

It seems like a shame to leave the website down for so long.  I know it was lost in the ubuntu server upgrade a year or two ago.  Several of the original pages (including the home page and the license page) are still accessible via the wayback machine.  If you'd like, I can reconstruct static versions of those pages, remove the forum and ticketing links and fix the other links to point to the current software and information.  If I did this, would you consider putting them back in place of the current text-only static page?  I mention this because when I recommend EmBitz to new users, it looks to them like the product is dead (which is not true).

BTW, I am looking forward to 2.0 and will happily pay for the upgrade!

Miinik, 2020-04-23:
I am also waiting for version 2.0, the announcement of which was six months ago.

I also think that it is necessary to restore the main page of the site and the forum. There are questions about EmBitz, but nowhere to ask them. A forum is needed.

AVI-crak, 2020-04-23:
I want to see the whole list of alien technologies for building IDEs at home. Or even a ready-made archive with all alien technologies, and a quick reference / translator from Martian to literary English.
It seems that there is a technology where the entire programming / code building environment can be placed in a separate container and run in the cloud or locally. The same settings for everyone who connected to the project, on any machine, from anywhere in the world. That would be perfect.

EmBitz, at the moment, the threshold for entering the project is very high.

EmBitz, 2020-04-25:
I don't have a clue if there is still need for a forum but I brought back the old one alive.
I don't have the categories any more so Dalbert if you have a screen shot then I will copy that.

https://www.embitz.org/forum

Miinik, 2020-04-25:
When registering, a letter does not come in email.

dalbert2, 2020-04-25:
Wow, that's great, I think the forum is hugely useful.

A few links from waybackmachine that show some top-level forums:

  * ARM toolchains
    <https://web.archive.org/web/20190107221307/https://www.embitz.org/forum/forum-8.html>
    › ArmGCC - EmBlocks - Armgcc from EmBlocks
    <https://web.archive.org/web/20190107221307/https://www.embitz.org/forum/forum-14.html>
    https://web.archive.org/web/20190107221307/https://www.embitz.org/forum/showthread.php?tid=32
  * ARM toolchains
    <https://web.archive.org/web/20190107221307/https://www.embitz.org/forum/forum-8.html>
    › ArmGCC - EmBlocks
    <https://web.archive.org/web/20190107221307/https://www.embitz.org/forum/forum-14-page-7.html>
    - How to build for minimal size
    https://web.archive.org/web/20190107221307/https://www.embitz.org/forum/showthread.php?tid=56
  * ARM toolchains
    <https://web.archive.org/web/20190107221306/https://www.embitz.org/forum/forum-8.html>
    › ArmGCC - EmBlocks
    <https://web.archive.org/web/20190107221306/https://www.embitz.org/forum/forum-14-page-7.html>
    - [solved]Same compiler, different IDE, different code/RAMsize
    https://web.archive.org/web/20190107221306/https://www.embitz.org/forum/showthread.php?tid=85
  * ARM toolchains-> ARM gdb - STLink V2.J20.S0 issues
    https://web.archive.org/web/20190107161008/https://www.embitz.org/forum/showthread.php?tid=310
  * EmBitz - Public
    <https://web.archive.org/web/20170817001444/https://www.embitz.org/forum/forum-16.html>
    › EmBitz IDE - Adapting STM32CubeMX Projects for EmBitz
    <https://web.archive.org/web/20170817001444/https://www.embitz.org/forum/forum-3.html>
    https://web.archive.org/web/20170817001444/https://www.embitz.org/forum/thread-670.html

EmBitz, 2020-04-25:
Email verification should work now. Login and ask for a resend.

EmBitz, 2020-04-27:
Forum is operational

closed 2020-04-27 by EmBitz