EmBitz/EBlink #22: source code

stappersg, 2020-07-18:
Where is the source code?

EmBitz, 2020-07-19:
Hoi Geert, op mijn server.

closed 2020-07-19 by EmBitz

stappersg, 2020-07-19:
> Any problems with the STM32L072 support?

Yes, but not big enough to bail out to binary blobs    :-)

EmBitz, 2020-07-20:
All the device related stuff is in EBlink done with c-like scripting so need for binary or even source code. 
It's all about the script.

stappersg, 2020-07-20:
On Mon, Jul 20, 2020 at 12:03:57AM -0700, EmBitz wrote:
> All the device related stuff is in EBlink done with c-like scripting so need for binary or even source code. 
> It's all about the script. 

And where to find the script and how the recognize it?

EmBitz, 2020-07-20:
In the /scripts folder of this repro perhaps?

I have the feeling that you are a bit lost here. Any experiences with bare metal development?

stappersg, 2020-07-21:
FWIW there is https://github.com/EmBitz/EBlink/blob/master/scripts/stmicro/stm32l0x.script

EmBitz, 2020-07-21:
If you need a GDB debugger or cli flash tool, just follow the readme.md and everything falls in place. It's so awful simple.

If you need source code to hack then I would suggest OpenOCD as starting point.

Miinik, 2020-07-22:
Didn't find descriptions of functions used in scripts. One has to guess what each of the functions is for by studying scripts. It would be great if there was a list of all the functions used in the scripts and their description.