EmBitz/EBlink #23: STM32WB55 support

fileshadow0rg, 2020-09-22:
EBlink version 2.9-[159] by Gerard Zagema

Interface at  2:6 - <BUS>:<ADDR>
Interface type : STlink/V2.1
STlink connect : Under reset
Target voltage : 3.28V
Interface speed: 4000KHz
Target detected: Cortex-M4 (r0p1) with FPv4_SP
HW breakpoints : 6
HW watchpoints : 4
Fault unwind   : Enabled
STmicro device : 0x495

Please report this ID so that we can add it.
Error unsupported STM32 ID: 0x495

EmBitz, 2020-09-22:
BTW if you know to which family it belongs then you can try to add it yourself.

Just add the device ID in the stm32-auto at the right place and also add it in the included device script on level deeper.

EmBitz, 2020-09-23:
[stm32wb55.zip](https://github.com/EmBitz/EBlink/files/5269656/stm32wb55.zip)

Hi,

- Replace **_stm32-auto.script_**
- Place **_stm32wb55xx.script_** in the stmicro folder
- Place  **_stm32wbxx_fl.scrip_**t in the stmicro\flash folder

Have a look at the scripts and you will see that it is very easy. The mass erase is not supported on CPU1 and if that script function is missing EBlink will emulate automatically mass erase by erasing all the sectors.

Good luck

closed 2020-09-23 by EmBitz