EmBitz/EBlink #24: EBlink V3.0 preview

Miinik, 2020-09-24:
From the context menu does not work.

[image]

EmBitz, 2020-09-25:
A shit, I forgot to set the environment variables. All my test systems had those already.

Thanks

EmBitz, 2020-09-25:
The EBlinkShell was also linked against dynamic  CRT dll's not available on any system. I rebuild it with static CRT lib's.
The Installer is replaced at 25 september.

The code is not yet signed with a certificate but that will be done in a couple of weeks.

Miinik, 2020-09-25:
I found a strange error.
If you start the GDB server from the menu, the name of the menu item changes to **EBlink exit**
If you click on this item, the Windows system shutdown dialog is displayed. Tested on Windows 8 and 10. 

If drivers are not installed for ST-Link, then EBlink reports an error.

[image]

It is necessary for the installer to check if there are drivers in the system.

EmBitz, 2020-09-25:
Yes, I saw that once at a text machine but never saw it again so I thought I was clicking something wrong.

Do you have that issue also after restart?

Drivers.... tsja I will think it over

Miinik, 2020-09-25:
Restarting Windows did not fix the error. Also, after this error, the Start button on the taskbar does not work. I click on it, but nothing happens (the menu is not displayed).

I forgot to write in the previous message. In Windows 10, the installer does not restart explorer. It exits, but it does not start. I had to manually start it.
In Windows 8, explorer restarts normally.

EmBitz, 2020-09-25:
You mean restart of explorer at uninstall?

EmBitz, 2020-09-25:
I can't reproduce it.

Miinik, 2020-09-25:
During install and uninstall in Windows 10.
I have installed and uninstalled several times and only one explorer has restarted. In other cases, black screen because the explorer is not running.

EmBitz, 2020-09-25:
Ok,I V3.0.1 

The problem is, EBlink can run as child in a parent process such as Explorer or Embitz or in Cmd.exe as a shell file. If the parent is cmd.exe we have to send close to this window. If the parent is not cmd.exe we have to send close directly to the EBlink window.

In the old situation I first tried EBlink window and if this didn't closed EBlink I closed the parent regardless who it was. This was a bit too much.

V3.0.1 checks first if the EBlink parent is cmd.exe and if so it will close the terminal, otherwise it will close EBlink window.

EmBitz, 2020-09-25:
btw The new installer will call the older uninstaller on request

EmBitz, 2020-09-25:
I tried it also with EBlink launched by Embitz and that goes perfect here

EmBitz, 2020-09-25:
V3.0.2 also skips the explorer restart at install time

EmBitz, 2020-09-25:
The reason why I close instead of terminate the process is because I will close STlink probe gracefully. You can check this with the status LED on the STlink probes.

Miinik, 2020-09-25:
OK, fixed all except restarting explorer in Windows 10 during uninstallation.
Thanks.

EmBitz, 2020-09-25:
Thanks Miinik,

Uninstall needs to restart explorer or otherwise it should ask for restart of windows.

A regsvr32 uninstall removes all the registries but the dll it self is still occupied by explorer hence not removable. So by restarting    explorer I can delete the installation directory completely inclusive the EBlinkShell.dll.

closed 2020-09-25 by EmBitz

EmBitz, 2020-09-25:
I installed windows 8.1 32 bit in virtual machine and I saw what you mean, explorer doesn't start up again.
V3.0.3 should tackle this.

Thanks again.

Miinik, 2020-09-25:
Excellent. There is no problem on Windows 10. Thanks.

By the way, the NSIS installer (if I'm not mistaken, you use it for EmBitz and EBlink) has multilingual support. It would be great if you add it.
I have attached to the post, a simple example installer with multilingual support.

[MultiLanguage.zip](https://github.com/EmBitz/EBlink/files/5284706/MultiLanguage.zip)

EmBitz, 2020-10-05:
@Miinik 
I looked at the multilingual but I'm not sure what it will bring us. If only the buttons are in the native language but everything else in English then it looks like a mesh I'm afraid. What is exactly what you are trying to achieve?

Miinik, 2020-10-05:
I tested the example from the archive and had all the text in the selected language, not just buttons.
Not all people English is their native language.

EmBitz, 2020-10-05:
I agree that English is not native to everybody, certainly not mine, but it's a sort of de facto standard in IT.
What I mean is that licence texts are still in English and the tool itself is still English, so I'm wondering what the added value would be, other than a lot of additional work for translations.

Miinik, 2020-10-06:
You don't need to translate. Need to add these lines to the installer script. 
```
;--------------------------------
;Languages

  !insertmacro MUI_LANGUAGE "English" ; The first language is the default language
  !insertmacro MUI_LANGUAGE "French"
  !insertmacro MUI_LANGUAGE "German"
  !insertmacro MUI_LANGUAGE "Spanish"
  !insertmacro MUI_LANGUAGE "SpanishInternational"
  !insertmacro MUI_LANGUAGE "SimpChinese"
  !insertmacro MUI_LANGUAGE "TradChinese"
  !insertmacro MUI_LANGUAGE "Japanese"
  !insertmacro MUI_LANGUAGE "Korean"
  !insertmacro MUI_LANGUAGE "Italian"
  !insertmacro MUI_LANGUAGE "Dutch"
  !insertmacro MUI_LANGUAGE "Danish"
  !insertmacro MUI_LANGUAGE "Swedish"
  !insertmacro MUI_LANGUAGE "Norwegian"
  !insertmacro MUI_LANGUAGE "NorwegianNynorsk"
  !insertmacro MUI_LANGUAGE "Finnish"
  !insertmacro MUI_LANGUAGE "Greek"
  !insertmacro MUI_LANGUAGE "Russian"
  !insertmacro MUI_LANGUAGE "Portuguese"
  !insertmacro MUI_LANGUAGE "PortugueseBR"
  !insertmacro MUI_LANGUAGE "Polish"
  !insertmacro MUI_LANGUAGE "Ukrainian"
  !insertmacro MUI_LANGUAGE "Czech"
  !insertmacro MUI_LANGUAGE "Slovak"
  !insertmacro MUI_LANGUAGE "Croatian"
  !insertmacro MUI_LANGUAGE "Bulgarian"
  !insertmacro MUI_LANGUAGE "Hungarian"
  !insertmacro MUI_LANGUAGE "Thai"
  !insertmacro MUI_LANGUAGE "Romanian"
  !insertmacro MUI_LANGUAGE "Latvian"
  !insertmacro MUI_LANGUAGE "Macedonian"
  !insertmacro MUI_LANGUAGE "Estonian"
  !insertmacro MUI_LANGUAGE "Turkish"
  !insertmacro MUI_LANGUAGE "Lithuanian"
  !insertmacro MUI_LANGUAGE "Slovenian"
  !insertmacro MUI_LANGUAGE "Serbian"
  !insertmacro MUI_LANGUAGE "SerbianLatin"
  !insertmacro MUI_LANGUAGE "Arabic"
  !insertmacro MUI_LANGUAGE "Farsi"
  !insertmacro MUI_LANGUAGE "Hebrew"
  !insertmacro MUI_LANGUAGE "Indonesian"
  !insertmacro MUI_LANGUAGE "Mongolian"
  !insertmacro MUI_LANGUAGE "Luxembourgish"
  !insertmacro MUI_LANGUAGE "Albanian"
  !insertmacro MUI_LANGUAGE "Breton"
  !insertmacro MUI_LANGUAGE "Belarusian"
  !insertmacro MUI_LANGUAGE "Icelandic"
  !insertmacro MUI_LANGUAGE "Malay"
  !insertmacro MUI_LANGUAGE "Bosnian"
  !insertmacro MUI_LANGUAGE "Kurdish"
  !insertmacro MUI_LANGUAGE "Irish"
  !insertmacro MUI_LANGUAGE "Uzbek"
  !insertmacro MUI_LANGUAGE "Galician"
  !insertmacro MUI_LANGUAGE "Afrikaans"
  !insertmacro MUI_LANGUAGE "Catalan"
  !insertmacro MUI_LANGUAGE "Esperanto"
  !insertmacro MUI_LANGUAGE "Asturian"
  !insertmacro MUI_LANGUAGE "Basque"
  !insertmacro MUI_LANGUAGE "Pashto"
  !insertmacro MUI_LANGUAGE "ScotsGaelic"
  !insertmacro MUI_LANGUAGE "Georgian"
  !insertmacro MUI_LANGUAGE "Vietnamese"
  !insertmacro MUI_LANGUAGE "Welsh"
  !insertmacro MUI_LANGUAGE "Armenian"
  !insertmacro MUI_LANGUAGE "Corsican"
  !insertmacro MUI_LANGUAGE "Tatar"
  !insertmacro MUI_LANGUAGE "Hindi"

;--------------------------------
;Reserve Files
  
  ;If you are using solid compression, files that are required before
  ;the actual installation should be stored first in the data block,
  ;because this will make your installer start faster.
  
  !insertmacro MUI_RESERVEFILE_LANGDLL
```
This will add support for major languages. For example, a multilingual installer compiled from my previous post. [MultiLanguage.zip](https://github.com/EmBitz/EBlink/files/5333565/MultiLanguage.zip)

The licensed text does not need to be translated. It can be selected in the window, copied to the clipboard and translated in Google translator or another.