EmBitz/EBlink #25: Source code missing

roberto-mainetti, 2020-10-04:
It's a very nice project, but on Github I think you should publish the source code, not the executable

EmBitz, 2020-10-05:
Yes, I feel what you mean and I also struggled with it. However, I saw other projects doing the same but perhaps I should change this,

This project doesn't focus on the binary code but on scripts. It doesn't try to be another OpenOCD, pyOCD or Texane. The purpose for github is/was sharing the scripts and modify them. The scripts are the assets. However, to give users an easy way to use it, it is convenient that the executables are in place. 

Perhaps this should be removed from github if it doesn't belong here.

Thanks!

closed 2020-10-05 by EmBitz

AVI-crak, 2020-10-05:
You cannot delete a part of a common project, otherwise you may end up with another fly in amber. How it happened with the IDE code. It seems to be there, but it is impossible to collect.

EmBitz, 2020-10-06:
But it is nice that you guys wants to contribute, why not modify script files?  E.g. the  stm32f3 and stm32f4  needs to be updated with dynamic memory mapping.