EmBitz/EBlink #26: EBlink v3.3 - errors in scripts

Miinik, 2020-10-05:
In the stm32gx.script file (and possibly in others) if there are errors in the names of the variables.
This is what happens when flashing the STM32G030 microcontroller.
[image]
```
Launching tool 'eblink': C:\Program Files (x86)\EmBitz\1.11\share\eblink\eblink.exe -I stlink,swd -S stm32-auto -a -F file=D:\STM32\G030\Test\bin\Release\Test.hex,verify,run (in C:\Program Files (x86)\EmBitz\1.11\share\eblink)
stderr> Script error :
stderr> SquirrelFunction<> call failed
stdout> 
stdout> EBlink version 3.3-[248] by Gerard Zagema
stdout> 
stdout> Interface USB# :
stdout> Interface type : STlink/V2
stdout> STlink connect : Under reset
stdout> Target voltage : 3.22V
stdout> Interface speed: 4000KHz
stdout> Target detected: Cortex-M0 (r0p1) 
stdout> HW breakpoints : 4
stdout> HW watchpoints : 2
stdout> Fault unwind   : Enabled
stdout> STmicro family : STM32G03/04xx
stdout> 
stdout> AN ERROR HAS OCCURED [getVarInfo: Could not retrieve UserData]
stdout> 
stdout> CALLSTACK
stdout> *FUNCTION [stm32_device()] stmicro/stm32gx.script line [84]
stdout> 
stdout> LOCALS
stdout> [sram_size] NULL
stdout> [cram_size] NULL
stdout> [page_size] 2048
stdout> [flash_size] 65536
stdout> [result] 0
stdout> [devApi] INSTANCE
stdout> [deviceId] 1126
stdout> [this] TABLE
Tool execution terminated with status -1
```
I fix this error for a previous version, but in the current version (3.3), does not have this fix. https://www.embitz.org/forum/thread-19-post-64.html#pid64

EmBitz, 2020-10-05:
Thanks a lot!
Strange, I tested it all weekend with G070 and G4 etc How this error came back, a mystery.

EmBitz, 2020-10-05:
I have updated the repro

EmBitz, 2020-10-06:
I changed it a bit, GDB can't handle zero length memory sections, I had problems with stm32G070.

I also notice a strange thing with that Nucleo-64 board. STutil can't connect the board (keeps trying), not even under reset but EBlink works just fine with it. Very strange.

closed 2020-10-06 by EmBitz