EmBitz/EBlink #27: Message: Please update probe to 16bit support Alternative

EmBitz, 2020-10-07:
If you flash a F3,F1 or F0 with an older STlink/V2 firmware then you can get a message to update the firmware. 

If you can't update your probe for any reason, there is a workaround available. In the stmicro/flash directory is a ***f3_f1_f0.script.alt*** script. Rename this to **f3_f1_f0.script** (rename the original e.g. ***f3_f1_f0.script.org***).

Background:
EBlink uses direct write to the MCU instead of external loaders in ram. The F3,F1 and F0 must be flashed with halfwords but the older versions of STlink/V2 didn't support this. If you update your firmware then everything is just fine. The alternative script is also detecting the missing feature of half words but will fallback to an external flash loader.

But I strongly recommend to update your probe firmware, but I think you already figured that out :)

Miinik, 2020-10-30:
You write that need to update the firmware in ST-Link, but on the main page you wrote **"Stop updating your stlink/v2 firmware if you want to keep using it for non-STmicro mcu's"**.
I think you should write to which version of the firmware is the best.

EmBitz, 2020-10-30:
Yes, as soon as I know more.
I'm not certain but some forums are mentioning  the same behaviour they already incorporated in the V3.

BuFran, 2021-01-31:
The stlink V2 firmware V2.J29.S7 doesn't work, always use firmware V2.J34.M25 and newer.