EmBitz/EBlink #28: EBlink: Flashing sector 4 failed

FELiXik, 2020-10-09:
Hello,
since v3.2 (v3.1 last working) I get error durring: Flashing sector  4, address: 0x08010000, length: 0x10000.

Interface at  1:2 - <BUS>:<ADDR>
Interface type : STlink/V2.1
STlink connect : Under reset
Target voltage : 2.90V
Interface speed: 4000KHz
Target detected: Cortex-M4 (r0p1) with FPv4_SP
HW breakpoints : 6
HW watchpoints : 4
Fault unwind   : Enabled
STmicro device : 0x413
Detected FLASH : 0x80000
Configured RAM : 0x30000

MCU - STM32F407VG.

Is there any solution?

Thank you and have a nice day!

EmBitz, 2020-10-09:
Try the current V3.6, a lot of changes since 3.2 both EBlink binary and EBlink scripts.
Please report if it's still there, thanks

FELiXik, 2020-10-09:
> Try the current V3.6, a lot of changes since 3.2 both EBlink binary and EBlink scripts.
> Please report if it's still there, thanks
> 
> P.s. both repro and installer are on 3.6 build 285

I tried also, still same error :(

EmBitz, 2020-10-09:
Yip, I found it. 
3.7 is on his way.

EmBitz, 2020-10-09:
This issue is called "the page size larger than 64K bug" 
Thanks for the report.

EmBitz, 2020-10-09:
Tested on win32 and Raspberry pi.

closed 2020-10-09 by EmBitz