EmBitz/EBlink #31: GDB read memory reply is repeated

erian747, 2020-11-18:
Im using EBlink version 3.8-[55] on linux (Ubuntu 20)
ST-link firmware version V2J33S7

I am starting a GDB server with: ./eblink -I stlink -S auto -P ../scripts/ -G nc

Connection to target works fine:
//------------------------------------------------------
EBlink version 3.8-[55] by Gerard Zagema

Interface USB# : 30FF6C064D50353343152443
Interface type : STlink/V2
STlink connect : Under reset
Target voltage : 3.23V
Interface speed: 4000KHz
Target detected: Cortex-M4 (r0p1) with FPv4_SP
HW breakpoints : 6
HW watchpoints : 4
Fault unwind   : Active break (level 2)
STmicro family : STM32F405/407/415/417
Detected FLASH : 0x100000
Configured RAM : 0x30000
GDB rom caching: Disabled
GDB listening (0.0.0.0 @ 2331) 

when i connect to the GDB server and send a read memory command: $m8000000,4#25
the first response look ok: 00080020#8a
also the second time i send the the same read memory command the response is ok

But the third time, the response looks like this:
00080020#8a$00080020#8a$00080020#8a$00080020#8a$00080020#8a$00080020#8a$00080020#8a$00080020#8a$00080020#8a$00080020#8a$00080020#8a$00080020#8a$00080020#8a$00080020#8a

I have tried both with a script and with nc with the same result, answer is repeated.
It seems like the answer is sent for every character given to the server

Is there some problem in the GDB server code, or have i missunderstod the RSP protocol?

EmBitz, 2020-11-18:
Thanks, I will look into it.

EmBitz, 2020-11-18:
Could it be not handling the ack?  The server expect and ACK but receives a letter so will echo the answer with every wrong ack.
BTW you can disable the ack algo.

EmBitz, 2020-11-18:
If you send an ACK (='+') on every receive then you get what you expected

[image]

closed 2020-11-18 by EmBitz

erian747, 2020-11-18:
Thanks for the guidance, it will try to read the documentation better next time
Really appriciate your work with EBlink by the way!