EmBitz/EBlink #32: getVarInfo: Could not retrieve UserData

erian747, 2020-11-20:
Im trying to interface with a STM32F4Discovery kit with it's onboard st-link
When i try to connect with ./eblink -v 8 -I stlink -S auto -P ../scripts/ -G nc  i get:

```
Interface      : STlink
Device found at bus 1:5
Interface USB# : 56FF6B064889495233081887
Interface at  1:5  - <BUS>:<ADDR>
Interface type : STlink/V2
STlink Frmware : V2.J34.S0
STlink connect : Under reset
Target voltage : 2.88V
Interface speed: 4000KHz
Target selected: Cortex-M
Target detected: Cortex-M4 (r0p1) with FPv4_SP
HW breakpoints : 6
HW watchpoints : 4
Fault unwind   : Active break (level 2)
Search script auto.script in path: ../scripts/ 
Script found: auto.script 
STLINK_SWD_AP_WDATA_ERROR 
Required script load: stm32-auto.script 
STmicro device : 0x411 
AN ERROR HAS OCCURED [getVarInfo: Could not retrieve UserData]

CALLSTACK
*FUNCTION [set_target()] stm32-auto.script line [117]

LOCALS
[deviceId] 1041
[this] TABLE
Script error :
SquirrelFunction<> call failed
```

**Note the STLINK_SWD_AP_WDATA_ERROR**

Just to compare, i ran a test with https://github.com/stlink-org/stlink, same system/setup and hardware
It seems to work ok, see log below

```
2020-11-20T08:22:21 DEBUG common.c: *** looking up stlink version
2020-11-20T08:22:21 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-11-20T08:22:21 DEBUG common.c: stlink pid     = 0x3748
2020-11-20T08:22:21 DEBUG common.c: stlink version = 0x2
2020-11-20T08:22:21 DEBUG common.c: jtag version   = 0x22
2020-11-20T08:22:21 DEBUG common.c: swim version   = 0x0
2020-11-20T08:22:21 DEBUG common.c:     notice: the firmware doesn't support a swim interface
2020-11-20T08:22:21 DEBUG common.c: stlink current mode: mass
2020-11-20T08:22:21 DEBUG usb.c: JTAG/SWD freq set to 0
2020-11-20T08:22:21 DEBUG common.c: *** set_swdclk ***
2020-11-20T08:22:21 DEBUG common.c: stlink current mode: mass
2020-11-20T08:22:21 DEBUG common.c: *** stlink_enter_swd_mode ***
2020-11-20T08:22:21 DEBUG common.c: *** stlink_jtag_reset ***
2020-11-20T08:22:21 DEBUG common.c: *** stlink_reset ***
2020-11-20T08:22:21 DEBUG common.c: *** stlink_write_debug32 0x05fa0004 to 0xe000ed0c
2020-11-20T08:22:21 DEBUG common.c: Loading device parameters....
2020-11-20T08:22:21 DEBUG common.c: *** stlink_core_id ***
2020-11-20T08:22:21 DEBUG common.c: core_id = 0x2ba01477
2020-11-20T08:22:21 DEBUG common.c: *** stlink_read_debug32 0x20006411 at 0xe0042000
2020-11-20T08:22:21 DEBUG common.c: *** stlink_read_debug32 0x410fc241 at 0xe000ed00
2020-11-20T08:22:21 DEBUG common.c: *** stlink_read_debug32 0x0400c000 at 0x1fff7a20
2020-11-20T08:22:21 INFO common.c: F4xx: 192 KiB SRAM, 1024 KiB flash in at least 16 KiB pages.
2020-11-20T08:22:21 DEBUG common.c: *** stlink_reset ***
2020-11-20T08:22:21 DEBUG common.c: *** stlink_write_debug32 0x05fa0004 to 0xe000ed0c
2020-11-20T08:22:21 DEBUG gdb-server.c: Chip ID is 0x00000413, Core ID is 0x2ba01477.
2020-11-20T08:22:21 INFO gdb-server.c: Listening at *:4242...
2020-11-20T08:24:26 DEBUG common.c: *** stlink_force_debug_mode ***
2020-11-20T08:24:26 DEBUG common.c: *** stlink_reset ***
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0x05fa0004 to 0xe000ed0c
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0x00000003 to 0xe0002000
2020-11-20T08:24:26 DEBUG common.c: *** stlink_read_debug32 0x00000261 at 0xe0002000
2020-11-20T08:24:26 INFO gdb-server.c: Found 6 hw breakpoint registers
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0002008
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe000200c
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0002010
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0002014
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0002018
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe000201c
2020-11-20T08:24:26 DEBUG gdb-server.c: init watchpoints
2020-11-20T08:24:26 DEBUG common.c: *** stlink_read_debug32 0x01000401 at 0xe000edfc
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0x01000401 to 0xe000edfc
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0001028
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0001038
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0001048
2020-11-20T08:24:26 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0001058
2020-11-20T08:24:26 INFO gdb-server.c: GDB connected.
2020-11-20T08:24:41 DEBUG gdb-server.c: recv: m20001000,1c
2020-11-20T08:24:41 DEBUG common.c: *** stlink_read_mem32 ***
2020-11-20T08:24:41 DEBUG gdb-server.c: send: 00000000000000000000000000000000000000000000000000000000
2020-11-20T08:24:42 DEBUG gdb-server.c: recv: m20001000,1c
2020-11-20T08:24:42 DEBUG common.c: *** stlink_read_mem32 ***
2020-11-20T08:24:42 DEBUG gdb-server.c: send: 00000000000000000000000000000000000000000000000000000000
2020-11-20T08:24:43 DEBUG gdb-server.c: recv: m20001000,1c
2020-11-20T08:24:43 DEBUG common.c: *** stlink_read_mem32 ***
2020-11-20T08:24:43 DEBUG gdb-server.c: send: 00000000000000000000000000000000000000000000000000000000
```

EmBitz, 2020-11-20:
Have you tried the latest script files from git?

EmBitz, 2020-11-20:
Ok found it, thanks!!!

EmBitz, 2020-11-20:
Or use this replacement script

[stm32-auto.zip](https://github.com/EmBitz/EBlink/files/5572555/stm32-auto.zip)

closed 2020-11-20 by EmBitz

EmBitz, 2020-11-20:
Fixed in repro and EBlinkInstaller