EmBitz/EBlink #33: Error: Connecting with target lost

erian747, 2020-11-20:
Now i got my STM32F4Discovery up and running if have encountered another problem :)

Im am polling the MCU once every second with: $m20001000,1c#b0 when target is running

It works for a while, but often in less than 2 to 3 minutes it somehow get disconnected from target, see log below
Im invoking with ./eblink -v 8 -I stlink,dr -S auto -P ../scripts/ -G nc

Same thing happens when i tested with a STM32F072-Disco board (Cortex-m0) and also with a custom board attached to a ST-Link V/2

Note that im running Linux/Ubuntu20 as a Virtualbox VM guest in a Win7 host, but never had any problem with USB on other devices

Is there something i can do/debug in order to get more detailed information about what is causing the problem?
```
EBlink version 3.8-[57] by Gerard Zagema

Interface      : STlink
Device found at bus 1:9
Interface USB# : 56FF6B064889495233081887
Interface at  1:9  - <BUS>:<ADDR>
Interface type : STlink/V2
STlink Frmware : V2.J34.S0
STlink connect : Hot plugging
Target voltage : 2.88V
Interface speed: 4000KHz
Target selected: Cortex-M
Target detected: Cortex-M4 (r0p1) with FPv4_SP
HW breakpoints : 6
HW watchpoints : 4
Fault unwind   : Active break (level 2)
Search script auto.script in path: ../scripts/
Script found: auto.script
STLINK_SWD_AP_WDATA_ERROR
Required script load: stm32-auto.script
STmicro device : 0x411
Required script load: stmicro/stm32f4x.script
STmicro family : STM32F405/407/415/417
Detected FLASH : 0x100000
Configured RAM : 0x30000
Required script load: stmicro/flash/f7_f4_f2.script
Required script load: stmicro/flash/errors.script
GDB rom caching: Disabled
GDB listening (0.0.0.0 @ 2331) 
Caching sector 0
Caching sector 1
Caching sector 2
Caching sector 3
Caching sector 4
Caching sector 5
Caching sector 6
Caching sector 7
Caching sector 8
Caching sector 9
Caching sector 10
Caching sector 11
GDB connected
GDB disconnected
GDB listening (0.0.0.0 @ 2331) 
GDB connected
Error: Connecting with target lost

```

erian747, 2020-11-20:
Directly afterwards i tried st-util without unplugging the STM32F072-Disco board and i got

```
st-util
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_VERSION
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_ENTER
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_READCOREID
2020-11-20T13:26:38 ERROR common.c: Failed to read core_id
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
```
So maybe the ST-Link got locked up in some way
After unplugging and re-connecting the board, it works again

erian747, 2020-11-20:
Tested the same scenario but running eblink.exe in windows instead of the linux guest and it seems stable for atlest 30minutes now
So i guess the problem is in the USB drivers or VirtualBox
I will try the same test on native linux / raspberry pi next week and see if that works

EmBitz, 2020-11-20:
Hi, 
We are using EBlink self for several years and we have test sites where it is running for weeks (ubuntu and Pi) without having any problems with stability. 

Some how the libusb hotplug register callback detects an USB remove I guess I your case I guess.

EmBitz, 2020-11-20:
It would be nice that you first check the native OSes before raising an issue.
If you need more detailed support I would suggest using https://www.embitz.org/forum

Thanks!

EmBitz, 2020-11-20:
Just installed a fresh Ubuntu 20 in VMware, works like a charm.
I close it for now, for any support please visit the embitz forum.

[image]

closed 2020-11-20 by EmBitz