EmBitz/EBlink #34: ST-link stuck just after reset command

BuFran, 2021-01-30:
The issue #33 seems still not fixed on real hardware. 

I am trying to use your eblink with CLion (because too much time spent waiting for new embitz). 
I have latest CLion and eblink installed, using default scripts.

When i run eblink many times using embitz/wine, it works fine:

```
EBlink version 3.8-[59] by Gerard Zagema

Interface      : STlink
Device found at bus 3:8
Interface USB# : 49FF6D064984525137401587
Interface at  3:8  - <BUS>:<ADDR>
Interface type : STlink/V2
STlink Frmware : V2.J29.S7
STlink connect : Under reset
Target voltage : 3.26V
Interface speed: 4000KHz
Target selected: Cortex-M
Target detected: Cortex-M0 (r0p0) 
HW breakpoints : 4
HW watchpoints : 2
Fault unwind   : Active break (level 2)
Search script stm32-auto.script in path: /opt/eblink/scripts
Script found: stm32-auto.script
STmicro device : 0x442
Required script load: stmicro/stm32f0x.script
STmicro family : STM32F09x
Detected FLASH : 0x20000
Configured RAM : 0x8000
Required script load: stmicro/flash/f3_f1_f0.script
Required script load: stmicro/flash/errors.script
GDB listening (0.0.0.0 @ 2331) 
Caching sector 0
Caching sector 1
Caching sector 2
Caching sector 3
Caching sector 4
Caching sector 5
Caching sector 6
Caching sector 7
...
```

BUT when i try to use CLion to connect through eblink, it fails. 

```
/home/bufran/bin/stm32gdb.sh

EBlink version 3.8-[59] by Gerard Zagema

Interface      : STlink
Device found at bus 3:9
Interface USB# : 49FF6D064984525137401587
Interface at  3:9  - <BUS>:<ADDR>
Interface type : STlink/V2
STlink Frmware : V2.J29.S7
STlink connect : Under reset
Target voltage : 3.26V
Interface speed: 4000KHz
Target selected: Cortex-M
Target detected: Cortex-M0 (r0p0) 
HW breakpoints : 4
HW watchpoints : 2
Fault unwind   : Active break (level 2)
Search script stm32-auto.script in path: /opt/eblink/scripts
Script found: stm32-auto.script
STmicro device : 0x442
Required script load: stmicro/stm32f0x.script
STmicro family : STM32F09x
Detected FLASH : 0x20000
Configured RAM : 0x8000
Required script load: stmicro/flash/f3_f1_f0.script
Required script load: stmicro/flash/errors.script
GDB listening (0.0.0.0 @ 2331) 
Caching sector 0
Caching sector 1
Caching sector 2
Caching sector 3
...
Caching sector 63
GDB connected
Unmodified data sector 0, addr 0x8000000, length 2048
Unmodified data sector 1, addr 0x8000800, length 2048
Unmodified data sector 2, addr 0x8001000, length 2048
Unmodified data sector 3, addr 0x8001800, length 2048
Unmodified data sector 4, addr 0x8002000, length 2048
Unmodified data sector 5, addr 0x8002800, length 2048
Unmodified data sector 6, addr 0x8003000, length 2048
Unmodified data sector 7, addr 0x8003800, length 1696
Unmodified data sector 7, addr 0x8003EA0, length 352
Unmodified data sector 8, addr 0x8004000, length 2048
Unmodified data sector 9, addr 0x8004800, length 2048
Unmodified data sector 10, addr 0x8005000, length 2048
Unmodified data sector 11, addr 0x8005800, length 2048
Unmodified data sector 12, addr 0x8006000, length 2048
Unmodified data sector 13, addr 0x8006800, length 2048
Unmodified data sector 14, addr 0x8007000, length 2048
Unmodified data sector 15, addr 0x8007800, length 1216
Unmodified data sector 15, addr 0x8007CC0, length 832
Unmodified data sector 16, addr 0x8008000, length 2048
Unmodified data sector 17, addr 0x8008800, length 2048
Unmodified data sector 18, addr 0x8009000, length 2048
Unmodified data sector 19, addr 0x8009800, length 2048
Unmodified data sector 20, addr 0x800A000, length 2048
Unmodified data sector 21, addr 0x800A800, length 2048
Unmodified data sector 22, addr 0x800B000, length 312
Unmodified data sector 22, addr 0x800B138, length 40
Unmodified data sector 22, addr 0x800B160, length 300
Reset: system
Debugger connected to 127.0.0.1:2331
Signal: SIGTRAP (Trace/breakpoint trap)
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
GDB disconnected
GDB listening (0.0.0.0 @ 2331) 
Caching sector 43

Process finished with exit code 0
GDB Server stopped, exit code 137
```

The ST-link is then garbaged and new reconnection through GDB fails. Even embitz/wine can't resurrect from this situation.

Seems that big difference is the "reset command", when i leave empty the "reset command" box in CLion, i can't reset board, BUT all functions work like a charm. When i write there "monitor reset 0" then the stlink interface hangs forever and needs the usb disconnect-reconnect to heal from this state.

After some digging, the situation is same using command line gdb.

Is there any way to gather more info to search which fault occured ?

Which version of STlink  firmware do you use for testing ?

Best regards, frantisek

BuFran, 2021-01-30:
When i try to uncheck "after load" checkbox, the fault is different but the result is same:

```
/opt/eblink/linux/eblink -I stlink -P /opt/eblink/scripts -S stm32-auto.script -v 8 -G port=2331

EBlink version 3.8-[59] by Gerard Zagema

Interface      : STlink
Device found at bus 3:4
Interface USB# : 49FF6D064984525137401587
Interface at  3:4  - <BUS>:<ADDR>
Interface type : STlink/V2
STlink Frmware : V2.J29.S7
STlink connect : Under reset
Target voltage : 3.26V
Interface speed: 4000KHz
Target selected: Cortex-M
Target detected: Cortex-M0 (r0p0) 
HW breakpoints : 4
HW watchpoints : 2
Fault unwind   : Active break (level 2)
Search script stm32-auto.script in path: /opt/eblink/scripts
Script found: stm32-auto.script
STmicro device : 0x442
Required script load: stmicro/stm32f0x.script
STmicro family : STM32F09x
Detected FLASH : 0x20000
Configured RAM : 0x8000
Required script load: stmicro/flash/f3_f1_f0.script
Required script load: stmicro/flash/errors.script
GDB listening (0.0.0.0 @ 2331) 
Caching sector 0
Caching sector 1
Caching sector 2
Caching sector 3
Caching sector 4
Caching sector 5
Caching sector 6
...
Caching sector 57
GDB connected
Unmodified data sector 0, addr 0x8000000, length 2048
Unmodified data sector 1, addr 0x8000800, length 2048
Unmodified data sector 2, addr 0x8001000, length 2048
Unmodified data sector 3, addr 0x8001800, length 2048
Unmodified data sector 4, addr 0x8002000, length 2048
Unmodified data sector 5, addr 0x8002800, length 2048
Unmodified data sector 6, addr 0x8003000, length 2048
Unmodified data sector 7, addr 0x8003800, length 1696
Unmodified data sector 7, addr 0x8003EA0, length 352
...
Unmodified data sector 22, addr 0x800B138, length 40
Unmodified data sector 22, addr 0x800B160, length 300
Debugger connected to 127.0.0.1:2331
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
[... 223 lines cut ...]
STLINK_SWD_DP_FAULT
STLINK_SWD_AP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_AP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_AP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_AP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_AP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_AP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_AP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_DP_FAULT
STLINK_SWD_AP_FAULT
...
```

EmBitz, 2021-01-31:
Hi,
Currently I use for a STM32F09 project V2.J34.M25.
The fault is coming from STlink so there is no more info. The Access Point has detected a fault.

Which OS and release are you using?
I will see if I can get a rig with the same OS on it  and check it with our stm32F091 project.

BuFran, 2021-01-31:
I am using Linux Mint 20 (Ulyana probably, i dont remember names, sorry)

The problem can be the firmware in STlink. I was hoped i have newest firmware, but you said that yours is newer so i need to dig how to get newer firmware there to be latest as possible. My stlink updater downloads only V2.J29.S7 and says that it is the latest one.

Seems the problem is connected with hard fault in my code that is handled differently in EmBitz and CLion. Some dump of communication between GUI and GDB and EBlink can help to dig where is the problem/difference but i don't know how to get it. 

Surprise is that hardfault handler is not called when I flash firmware with Embitz, but it is there when i flash it (same elf file) with CLion or command line gdb.

Some progress has been done by installing ```gdb-multiarch``` package, not the default one on Debian.  with this multiarch gdb, the hardfaults are sporadically called, i can flash, debug the software, everything works (even i can debug those hardfaults in handler) but stlink crashes every time i exit gdb, and seconnd connect is errorneous (message target power perhaps) i need to disconnect/reconnect stlink to USB to workaround.

I am using StlinkV2 USB dongle and powering 3V3 from the application. I don't know how the power supply in StlinkV2 is made, but after USB disconnect, the stlink 3V3 output is powered from my application. I deduct there that stlink is not reseted (3v3 is always there) but USB layer is reinitialized by reconnecting to USB port.

EmBitz, 2021-01-31:
V2.J34.M25 is also not latest I believe and is already 2 years old but I never have problems with this one.

BuFran, 2021-01-31:
My versions on PC:

```
bufran@LTR-BuFran:~$ uname -a
Linux LTR-BuFran 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
bufran@LTR-BuFran:~$ gdb-multiarch --version
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
bufran@LTR-BuFran:~$ gdb --version
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

surprise, seems that installing gdb-multiarch replaced the original one.

I have raspi there V1 only, with very old raspbian only console (no X), i don't think there will be move forward in trying to install all compiler dependencies and making everything to work.

At first i'll try to find newest stlink firmware and update it. I have there more stlinks with different firmwares so i can test the differences later.

BuFran, 2021-01-31:
PROBLEM SOLVED

Updater said many times error response 0x01 before flash (lot of prblems to move into DFU mode) but after succesful flash of firmware, there is everything working without need to reconnect usb !

After second reconnect, the log is now:

```
/opt/eblink/linux/eblink -I stlink -P /opt/eblink/scripts -S stm32-auto.script -v 8 -G port=2331

EBlink version 3.8-[59] by Gerard Zagema

Interface      : STlink
Device found at bus 3:29
Interface USB# : 49FF6D064984525137401587
Interface at  3:29 - <BUS>:<ADDR>
Interface type : STlink/V2
STlink Frmware : V2.J37.S7
STlink connect : Under reset
Target voltage : 3.25V
Interface speed: 4000KHz
Target selected: Cortex-M
Target detected: Cortex-M0 (r0p0) 
HW breakpoints : 4
HW watchpoints : 2
Fault unwind   : Active break (level 2)
Search script stm32-auto.script in path: /opt/eblink/scripts
Script found: stm32-auto.script
STmicro device : 0x442
Required script load: stmicro/stm32f0x.script
STmicro family : STM32F09x
Detected FLASH : 0x20000
Configured RAM : 0x8000
Required script load: stmicro/flash/f3_f1_f0.script
Required script load: stmicro/flash/errors.script
GDB listening (0.0.0.0 @ 2331) 
Caching sector 0
Caching sector 1
Caching sector 2
....
```

My fault was that i think stlink __downloads__ newer firmware, but the truth is that flashing tool __holds__ an fixed version in file and newer firmware must be downloaded from the ST's web.

This is for others they dig same problem. 

Issue can be closed (or if you want some more info with older unflashed stlink, i can give it to you, i have an bunch of unflashed originals there and i can find the version which corrects the bug)

EmBitz, 2021-01-31:
Nice!

What is wise? Fix it for older firmware versions or build-in a warning error in the STlink interface driver module if the firmware version is too old. I'm not sure yet. Do you still have that STutil file with that older problematic firmware? I would like to have it.

BTW
I always use Rasp in console and never in X. It's very handy as a wireless debug probe setup.

BuFran, 2021-01-31:
Yes i have ST-LinkUpgrade / STUtil on windows machine installed. I can send you the package.

EmBitz, 2021-01-31:
That would be awful nice.
Do you have by any chance a https://embitz.org/forum account? If so, you could PM me the exe in a zip file.

Cheers

BuFran, 2021-01-31:
[STM32 ST-LINK Utility.zip](https://github.com/EmBitz/EBlink/files/5899503/STM32.ST-LINK.Utility.zip)

There is my original STlink V2.J29.S7

EmBitz, 2021-01-31:
Much easier, mucha gracias

closed 2021-01-31 by EmBitz

BuFran, 2021-01-31:
Yay yes i have paid embitz in the past but i haven't there for a long time. :o)

EmBitz, 2021-01-31:
I forgive you :)

BuFran, 2021-02-01:
So I have bad news. The problem is still there. It is now not connected with reset, but with long time of connection over usb.

I gathered new info.

```
(gdb) show config
This GDB was configured as follows:
   configure --host=x86_64-pc-linux-gnu --target=x86_64-pc-linux-gnu
             --with-auto-load-dir=$debugdir:$datadir/auto-load
             --with-auto-load-safe-path=$debugdir:$datadir/auto-load
             --with-expat
             --with-gdb-datadir=/linux/share/gdb (relocatable)
             --with-jit-reader-dir=/linux/lib/gdb (relocatable)
             --without-libunwind-ia64
             --with-lzma
             --without-babeltrace
		             --with-intel-pt
             --without-mpfr
             --without-xxhash
             --with-python=/linux (relocatable)
             --without-guile
             --disable-source-highlight
             --with-separate-debug-dir=/usr/lib/debug
             --with-system-gdbinit=/etc/gdb/gdbinit
```

and gdb config file
```
bufran@LTR-BuFran:~$ cat /etc/gdb/gdbinit
# System-wide GDB initialization file.
```

seems the root cause is that multiarch gdb doesn't work well

Can you please tell which gdb you are using ?

EmBitz, 2021-02-01:
7.8.2
But I tailored it heavily with additional query parameters and new (missing) commands.
I should upgrade it but it is still running fine for me, I have a lot of other things on my list.

BuFran, 2021-02-01:
I have figured the command that always stuck the gdb:

```
(gdb) load
`/data/21/PCBS/LTR1919/fw/bin/LTR1919.elf' has changed; re-reading symbols.
Loading section .text, size 0xba80 lma 0x8000000
Loading section .init_array, size 0x20 lma 0x800ba80
Loading section .ARM.exidx, size 0x8 lma 0x800baa0
Loading section .data, size 0x15c lma 0x800baa8
Start address 0x08000ed4, load size 48132
Transfer rate: 20 KB/sec, 8022 bytes/write.

(gdb) mon reset 0
(gdb) cont
Continuing.
```

and now the gdb doesn't respond to "pause" button in ide.

output from eblink is

```
...
Flashing sector 21, address: 0x0800A800, length: 0x0800
Erasing  sector 22
Flashing sector 22, address: 0x0800B000, length: 0x0800
Erasing  sector 23
Flashing sector 23, address: 0x0800B800, length: 0x0400

Reset: system
Command timed out
GDB disconnected
GDB listening (0.0.0.0 @ 2331) 
Caching sector 0

Process finished with exit code 0
GDB Server stopped, exit code 137
```

__Command timed out__ ... first reasonable output !

BuFran, 2021-02-01:
gdb 7.8.2 multiarch or arm-none-eabi-gdb ?

EmBitz, 2021-02-01:
arm-none-eabi-gdb

EmBitz, 2021-02-01:
(gdb) show config
This GDB was configured as follows:
   configure --host=i686-pc-mingw32 --target=arm-none-eabi
             --with-auto-load-dir=$debugdir:$datadir/auto-load
             --with-auto-load-safe-path=$debugdir:$datadir/auto-load
             --with-expat
             --with-gdb-datadir=/em_gdb/arm/share/gdb (relocatable)
             --with-jit-reader-dir=/em_gdb/arm/lib/gdb (relocatable)
             --without-libunwind-ia64
             --without-lzma
             --without-guile
             --with-separate-debug-dir=/em_gdb/arm/lib/debug (relocatable)
             --without-zlib
             --without-babeltrace

BuFran, 2021-02-05:
I'm now traversing through output of gdb, and digging where is the problem. 

```set debug target 1``` reveals nothing usable with target stlink but one warning:

```warning: while parsing target description (at line 1): Target description specified unknown architecture "arm"```

Seems that finally multiarch gdb is not compatible with ARM. On the first sight it works but this issue reveals that it doesn't work fluently.

in the log:

```
(gdb) set debug target 1
(gdb) tar remote localhost:2331
-> exec->log_command (...)
<- exec->log_command (tar remote localhost:2331)
-> remote->open (...)
target_close ()
Remote debugging using localhost:2331
remote:target_xfer_partial (10, target.xml, 0x55ec52be9d10, 0x0, 0x0, 4096) = 1, 970, bytes =
 3c 3f 78 6d 6c 20 76 65 72 73 69 6f 6e 3d 22 31 ...
remote:target_xfer_partial (10, target.xml, 0x55ec52beb0ea, 0x0, 0x3ca, 4096) = 0, 0
warning: while parsing target description (at line 1): Target description specified unknown architecture "arm"
warning: Could not load XML target description; ignoring
remote:target_xfer_partial (6, (null), 0x55ec52bead20, 0x0, 0x0, 4096) = -1, 0
target_fetch_registers (eip) = ffffffff 0xffffffff 4294967295
```

```eip``` is x86 register so this is probably hit to incompatibility of my GDB with ARM.

I will try to install your's version of GDB and give you an reply if it worked.

BuFran, 2021-02-05:
:-(

```
/opt/eblink/linux/eblink -I stlink -P /opt/eblink/scripts -S stm32-auto.script -G port=2331 -v 8

EBlink version 3.8-[59] by Gerard Zagema

Interface      : STlink
Device found at bus 3:30
Interface USB# : 49FF6D064984525137401587
Interface at  3:30 - <BUS>:<ADDR>
Interface type : STlink/V2
Error during pretty printers setup: Undefined info command: "pretty-printer".  Try "help info".

Some features and performance optimizations will not be available.

STlink connect : connect under reset failed, try without reset
Error initializing interface stlink
Target power perhaps?

GDB Server stopped, exit code 255

Process finished with exit code 0
```

disconnecting usb and reconnecting helps a lot - it works fine, until i exit gdb/eblink session. Next session always garbaged.

```
bufran@LTR-BuFran:~$ arm-none-eabi-gdb -version
GNU gdb (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.1.90.20201028-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

The problem is still not found where it is. Possibilities:

 * ---Bug in stlink firmware--- (i have newest, you tested it)
 * Bug in eblink incorrectly starting stopping between sessions
 * ---Bug in eblink version--- (i have newest, you tested it)
 * Bug in gdb incorrectly starting stopping between sessions
 * ---Bug in gdb version--- (bug is present and same on more versions of gdb)
 * Bug in CLion sending surprise commands to gdb

I will investigate now:

 * If it still be over arm-none-eabi-gdb (pure commandline) to opt-out CLion bug
 * To get exact report of ```set debug target 1``` when this issue fires.
 * When it doesn't fire, it should be CLion's bug.

BuFran, 2021-02-05:
```
Sending packet: $D#44...Packet received: 
Sending packet: $vKill;a410#33...Packet received: 
Packet vKill (kill) is NOT supported
Sending packet: $k#6b...
```

This is last message just before EBlink (GDB server if I understand) ends with exit code 137 (OutOfMemory) and next reconnect fails with 

```
Error initializing interface stlink
Target power perhaps?
```

I don't understand the debug probe architecture, so the ```vKill``` is function of ST-link, EBlink, or GDB ?

BuFran, 2021-02-05:
BUT seems that same things  from commandline arm-none-eabi-gdb everything works fine, i can send multiple kills from gdb, start stop gdb, restarting eblink in different orders and no other error happens. 

Surprise.

The CLion should handle the connection different way, probably killing eblink with EOutOfMemory signal (exit code 137)

HIT !

PROBLEM SOLVED by adding parameter ```-G s``` to the eblink commandline. 

The Clion actually kills the running gdbserver if it stays open after detaching the gdb. And the
signal routine in EBlink is not correctly handled so the STlink firmware is garbaged and can't
be reinitialized on next GDBserver run without USB reconnect. 

Just for others who walk with same problem :-)

EmBitz, 2021-02-05:
Great!

The SIGINT and SIGABRT are handled by Linux EBlink. 
But actively closing often gives problems so you can better let EBlink quit on connection close as you already discovered. That's why the 's' option is there. It will give you a fastest clean close as possible after GDB is done.

Making an IDE working nicely with GDB and GDB server is mostly a matter of fine tuning. You start with some minor fixes and before you know, you are building your own IDE :)

Cheers.