EmBitz/EBlink #37: STM32L45/L46xx DeviceID 0x462 with 160k SRAM

sneuf, 2021-09-24:
Please change line 86 in stm32l4x.script to ram_size = 0x28000 to get it work for DeviceID 0x462 STM32L45/L64xx with 160k SRAM.

EmBitz, 2021-09-25:
This is something that I'm struggled with in the past how to implement this because we can't get the amount of RAM easily from the CPU registers like the FLASH size.

It's nice to have the exact amount of RAM so that GDB will give meaningful feedback if your are accessing non existing RAM. So I'm on the safe side and you can override the amount of RAM on the command line with -D RAM_SIZE=0x28000.

But during this typing an idea is popping up, perhaps we can probe certain memory locations to check the ram size. That's the power of that scripting engine. I will do some tests as soon as I have some spare time...

sneuf, 2021-09-25:
Thank you for the command line arguments tip. I don't found this option by myself in the EBlink description. But this is completely sufficient for my needs.

I suggest to define in the scripts the maximum values per DeviceID for FLASH and SRAM size and to add the -D RAM_SIZE and FLASH_SIZE command line arguments clearly as examples to the EBlink description.

So out of the box all variants within a DeviceID will work and if you want to have the limited values of a variant be present in GDB, user can configure it in the EmBitz EBlink settings or command line arguments.

In my case this is for STM32L451RE: 512k FLASH and 160k SRAM:

[image]

EmBitz, 2021-09-25:
A lot of features are unknown to regular users due to lack of documentation. There are a lot of nice things possible with EBlink.

EmBitz will get a totally new GDB driver where EBlink is tightly integrated: 
[image]

EmBitz, 2021-09-25:
Btw:

You can also define your own optional script variables. E.g. the part that makes RAM_SIZE possible is pure scripting and is:

`    // The user specified the size of ram memory
    if (isScriptObject("RAM_SIZE") && RAM_SIZE>0)
    {
      ram_size = (RAM_SIZE & 0xffff) * 1024
      printf("CLI set    RAM : 0x%X\n", ram_size)
    }`

sneuf, 2021-09-25:
Great thing with the new EmBitz EBlink integration. And a very good job of all EmBitz/EBlink programmers!

We really look forward to the release day of the new EmBitz.

So if you need help with review, testing, documentation, add new devices and so on please give as a hint.

closed 2021-09-26 by EmBitz