EmBitz/EBlink #39: EBlink install overwrites PATH

frank-freihube-rmcan-com, 2021-11-05:
When installing EBlink the path to the executable gets written to PATH with replace instead of append.

Yesterday I installed EmBitz 2.0 with EBlink and after the installation the only thing inside PATH (which got also renamed to `Path`) is now
```
C:\Program Files (x86)\EBlink\win32
```

EmBitz, 2021-11-07:
That's very strange because the installation script uses append.
Also, there are more than 600 installations and you are the first with this issue.

Perhaps this could be a OS issue. A couple of months ago I did a windows update and had the same issue but not with EBlink but with some microsoft as the sole path entry.

EmBitz, 2021-11-07:
Did you had perhaps a very long path string?

EmBitz, 2021-11-07:
I used the EnVarUpdate which is not right for paths if they contain large number of paths.
I changed it in the current installer to EnVar::AddValue

Thanks for the report and sorry for the troubles. I know how terrible annoying this is.

closed 2021-11-07 by EmBitz

frank-freihube-rmcan-com, 2021-11-08:
Thanks for the quick response, this probably has been the issues as my path is indeed quite long. After setting everything up it is now 955 chars again and probably has been longer before as I probably still forget to add some things^^